e2nIEE/pandapower #944: Dependencies for Juila, PowerModels and all associated packages (JuMP, Ipopt, Cbc etc)

Jholweger, 2020-10-08:
Hello Community! 

I'm having a lot of trouble making pandapower working with PowerModels.jl. In particular, in the notebook [storage_powermodels.ipynb](https://github.com/e2nIEE/pandapower/blob/develop/tutorials/storage_powermodels.ipynb), I wasn't able to call `pp.runpm_storage_opf`without getting a `KeyError "p_loss" not in key` from PowerModels. 

Later on, I figured out that this `p_loss`  was introduced in power models after v0.12.5. So I downgraded PowerModels to this version. Now the later notebook works fine but, going back to a standard case in [opf_powermodels.ipynb](https://github.com/e2nIEE/pandapower/blob/develop/tutorials/opf_powermodels.ipynb), I am now getting an error from JuMP.jl call from [pp_2_pm.jl](https://github.com/e2nIEE/pandapower/blob/develop/pandapower/opf/pp_2_pm.jl) saying that `optimizer_with_attributes` is  not defined. Again checking, this method was implemented after JuMP v0.21.0 which is **not compatible** with PowerModels 0.12.5!!!! 

Would it be possible to specify the dependencies for running PowerModels with Julia in a specific way?? 
Anyone happy to share his working environment?

friederikemeier, 2020-10-14:
I recently worked with Powermodels 0.17.3. Try that one, all tests were passing with that version. I need to document this better, sorry!

rbolgaryn, 2020-11-06:
No updates anymore, closing. @Jholweger feel free to reopen if you have additional questions or input.

closed 2020-11-06 by rbolgaryn