e2nIEE/pandapower #946: Query about transformer

NontasPapapadopoulos, 2020-10-08:
Hello gentlemen, 

I'm trying to create a transformer with specific values of resistance, reactance and susceptance (pu values). 
Could you explain me how to do this? 

And in reverse, I am wondering if there is a way to know the resistance, reactance and susceptance of the transformer in per unit. 

Thank you in advance!

Marahop, 2020-10-09:
Hello everyone, 

I would like to know the same. I would like to being able to say if the load of the transformer is 50% then the Energy output will be x and if it is 100% it is y. Happy for any advice,
Thank you very much.

lthurner, 2020-10-19:
> I'm trying to create a transformer with specific values of resistance, reactance and susceptance (pu values).
Could you explain me how to do this?

Check out the code of the PYPOWER / MATPOWER converter that is included in pandapower, such a transformation is applied there. You will need the rated apparent power of the transformer however, otherwise you can't calculate the values.

> And in reverse, I am wondering if there is a way to know the resistance, reactance and susceptance of the transformer in per unit.

The easiest way is to run a power flow and check the internal values in the ppc. There is a tutorial about internal data structures that you can look into for how to do this.

lthurner, 2020-10-19:
> I would like to know the same. I would like to being able to say if the load of the transformer is 50% then the Energy output will be x and if it is 100% it is y. Happy for any advice,

This seems like a different issue, I am not sure what you mean with energy output? The power flow calculates only power, not energy values.

rbolgaryn, 2020-10-20:
Also, loading of branch elements in pandapower is calculated based on their current divided by their rated current.

NontasPapapadopoulos, 2020-10-20:
> The easiest way is to run a power flow and check the internal values in the ppc. There is a tutorial about internal data structures that you can look into for how to do this.

could you give me a link please?

rbolgaryn, 2020-10-21:
> could you give me a link please?

It's in your pandapower installation:

pandapower\tutorials\internal_datastructure.ipynb

rbolgaryn, 2020-11-06:
This seems to be also interesting in the context of the issue #962. The functionality to calculate per-unit values can be expanded to branch parameters (can be considered at least).

closed 2020-11-06 by rbolgaryn