e2nIEE/pandapower #947: Classical Model of Synch. Gen for power flow

akshay-rachakonda, 2020-10-10:
For power flow we use classical model of synch gen. which is constant voltage source behind steady state reactance. In pandapower function there is no option  for including steady state reactance

pandapower.create_gen(net, bus, p_mw, vm_pu=1.0, sn_mva=nan, name=None, index=None, max_q_mvar=nan, min_q_mvar=nan, min_p_mw=nan, max_p_mw=nan, min_vm_pu=nan, max_vm_pu=nan, scaling=1.0, type=None, slack=False, controllable=nan, vn_kv=nan, xdss_pu=nan, rdss_pu=nan, cos_phi=nan, in_service=True).

How can I include the reactance.

rbolgaryn, 2020-10-13:
Hi @akshay-rachakonda ,

do you need it for the load flow study or the short circuit study?
Have you considered using the element "impedance"? Do you think creating an impedance element with the generator in combination would solve it?
Also, it sounds similar to the extended ward element: https://pandapower.readthedocs.io/en/v1.4.1/elements/xward.html
It has a voltage source, a series impedance and a shunt impedance. Can you modify the parameters in a way that represents your particular model?

Roman

akshay-rachakonda, 2020-10-14:
I want it for power flow study.

Issue with impedence function:
1. Impedence is placed between two buses. But in my case it is between a generator and a bus.

Issue with extended ward
1. This function is for load buses (pq) buses and I am working on generator (pv) bus so I am not sure how to proceed.

rbolgaryn, 2020-10-14:
Okay, I see. 
I think the easiest solution for you would be to create auxiliary buses at generators, and then add the impedance elements between the auxiliary buses and the generator connection buses in the grid.

rbolgaryn, 2020-10-14:
Adding this functionality to the gen element would depend on how important it is/how often it is needed. If you want, you are welcome to submit a pull request that implements the reactance of the synchronous generators in pandapower internally, without the need to create an additional element.

akshay-rachakonda, 2020-10-14:
@rbolgaryn thanks roman

akshay-rachakonda, 2020-10-14:
@rbolgaryn  for now I have added synch impedence to line impedance.