e2nIEE/pandapower #948: Load FLow Bug

Homam94, 2020-10-11:
Hello everybody,
I carry out a sensitivity analysis in my code, a load flow calculation has to be carried out when the generator feed is increased. After a few runs I get this error message, pandapower.powerflow.LoadflowNotConverged: Power Flow nr did not converge after 10 iterations!, which I could not fix with the help of diagnostic_ tool. Do you have any idea to fix this bug?

rbolgaryn, 2020-10-13:
Hi @Homam94 ,

it seems like the load flow did not converge. 
You can try calculating the sensitivities by using a smaller increment, maybe this will help.
Also, if you are adjusting them one-by-one, make sure that such adjustments are set back every time, before you go to the next adjustment.

Roman

lthurner, 2020-10-19:
If you keep increasing the generator feed-in, it is logical that you will at some point arrive at the point of infeasibility, where the power grid becomes unstable. Do you have any indication that this is a bug in pandapower, and you are not just reaching the physical limits of the system?

lthurner, 2020-10-29:
Closed - please reopen if you have further questions.

closed 2020-10-29 by lthurner