e2nIEE/pandapower #950: pp.runpm_storage_opf

bmitrauncc, 2020-10-13:
When I run storage_results = pp.runpm_storage_opf(net, n_timesteps=96, time_elapsed=0.25), I get an error..

FileNotFoundError                         Traceback (most recent call last)
<ipython-input-4-c4e37dec743b> in <module>
      6 # run the PowerModels.jl optimization
      7 # n_time steps = 96 and time_elapsed is a quarter of an hour (since the time series are in 15min resolution)
----> 8 storage_results = pp.runpm_storage_opf(net, n_timesteps=96, time_elapsed=0.25)

C:\ProgramData\Anaconda3\lib\site-packages\pandapower\runpm.py in runpm_storage_opf(net, calculate_voltage_angles, trafo_model, delta, trafo3w_losses, check_connectivity, n_timesteps, time_elapsed, correct_pm_network_data, pm_model, pm_time_limits, pm_log_level)
    359     net._options["time_elapsed"] = time_elapsed
    360 
--> 361     _runpm(net)
    362     storage_results = read_pm_storage_results(net)
    363     return storage_results

C:\ProgramData\Anaconda3\lib\site-packages\pandapower\opf\run_powermodels.py in _runpm(net, delete_buffer_file)
     34     buffer_file = dump_pm_json(pm)
     35     # run power models optimization in julia
---> 36     result_pm = _call_powermodels(buffer_file, net._options["julia_file"])
     37     # read results and write back to net
     38     read_pm_results_to_net(net, ppc, ppci, result_pm)

C:\ProgramData\Anaconda3\lib\site-packages\pandapower\opf\run_powermodels.py in _call_powermodels(buffer_file, julia_file)
     46     try:
     47         import julia
---> 48         from julia import Main
     49     except ImportError:
     50         raise ImportError("Please install pyjulia to run pandapower with PowerModels.jl")

C:\ProgramData\Anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

C:\ProgramData\Anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

C:\ProgramData\Anaconda3\lib\importlib\_bootstrap.py in _load_unlocked(spec)

C:\ProgramData\Anaconda3\lib\importlib\_bootstrap.py in _load_backward_compatible(spec)

C:\ProgramData\Anaconda3\lib\site-packages\julia\core.py in load_module(self, fullname)
    245         if juliapath == 'Main':
    246             return sys.modules.setdefault(fullname,
--> 247                                           JuliaMainModule(self, fullname))
    248         elif self.julia.isafunction(juliapath):
    249             return self.julia.eval(juliapath)

C:\ProgramData\Anaconda3\lib\site-packages\julia\core.py in __init__(self, loader, *args, **kwargs)
    147     def __init__(self, loader, *args, **kwargs):
    148         super(JuliaModule, self).__init__(*args, **kwargs)
--> 149         self._julia = loader.julia
    150         self.__loader__ = loader
    151 

C:\ProgramData\Anaconda3\lib\site-packages\julia\core.py in julia(self)
    237     @property
    238     def julia(self):
--> 239         self.__class__.julia = julia = Julia()
    240         return julia
    241 

C:\ProgramData\Anaconda3\lib\site-packages\julia\core.py in __init__(self, init_julia, jl_init_path, runtime, jl_runtime_path, debug, **julia_options)
    466             self.api = get_libjulia()
    467         elif init_julia:
--> 468             jlinfo = JuliaInfo.load(runtime)
    469             if jlinfo.version_info < (0, 7):
    470                 raise RuntimeError("PyJulia does not support Julia < 0.7 anymore")

C:\ProgramData\Anaconda3\lib\site-packages\julia\juliainfo.py in load(cls, julia, **popen_kwargs)
     71             stderr=subprocess.PIPE,
     72             universal_newlines=True,
---> 73             **popen_kwargs
     74         )
     75 

C:\ProgramData\Anaconda3\lib\subprocess.py in __init__(self, args, bufsize, executable, stdin, stdout, stderr, preexec_fn, close_fds, shell, cwd, env, universal_newlines, startupinfo, creationflags, restore_signals, start_new_session, pass_fds, encoding, errors, text)
    798                                 c2pread, c2pwrite,
    799                                 errread, errwrite,
--> 800                                 restore_signals, start_new_session)
    801         except:
    802             # Cleanup if the child failed starting.

C:\ProgramData\Anaconda3\lib\subprocess.py in _execute_child(self, args, executable, preexec_fn, close_fds, pass_fds, cwd, env, startupinfo, creationflags, shell, p2cread, p2cwrite, c2pread, c2pwrite, errread, errwrite, unused_restore_signals, unused_start_new_session)
   1205                                          env,
   1206                                          os.fspath(cwd) if cwd is not None else None,
-> 1207                                          startupinfo)
   1208             finally:
   1209                 # Child is launched. Close the parent's copy of those pipe

FileNotFoundError: [WinError 2] The system cannot find the file specified

lthurner, 2020-10-20:
This test needs the julia interface to be installed and configured properly (see: https://pandapower.readthedocs.io/en/v2.4.0/opf/powermodels.html#)

bmitrauncc, 2020-10-23:
Thanks for the workaround..

When I run it from anaconda prompt the module seems to work, but if I try to use it from spyder or inside a virtual environment it fails.. Do I need to install it in the specific environment?

ENV["<ENV NAME>"]="C:\Users\Bhaskar\Anaconda3\python.exe" ?

rbolgaryn, 2020-11-06:
@MMajidi137 can you take a look please?

MMajidi137, 2020-11-09:
> Thanks for the workaround..
> 
> When I run it from anaconda prompt the module seems to work, but if I try to use it from spyder or inside a virtual environment it fails.. Do I need to install it in the specific environment?
> 
> ENV[""]="C:\Users\Bhaskar\Anaconda3\python.exe" ?

You need to install spyder separate from anaconda. unfortunately, PyJulia cannot work well in anaconda https://pyjulia.readthedocs.io/en/latest/

rbolgaryn, 2020-12-02:
Hi @bmitrauncc , is the issue solved for you or is it still open?

SteffenMeinecke, 2021-11-11:
closed due to inactivity

closed 2021-11-11 by SteffenMeinecke