e2nIEE/pandapower #952: Error: "Estimation failed! Pandapower network failed to update!"

fmpnsc, 2020-10-13:
Hi there everyone,
I'm currently trying to implement a python script where I call the state estimation function during a "while" loop, so that I can recursively do the state estimation for every time period of data I posess.
Basically after doing some calculations on a specific dataframe ("df_opf_results") and using those results as values for the state estimation measurements (using the function "pp.create_measurement") I call the state estimation, save the results and start a new while iteration.
I'll leave the code below so everyone can see.
When I print the "net_1.res_bus_est" and "net_1.res_line_est" they both are empty for every iteration of the while loop.
At the end, the error "Estimation failed! Pandapower network failed to update!" appears. 

I would highly appreciate if anyone could help me in this problem since my Master's Thesis is currently in a stand-by situation because of this.
You can reach out to me at any time:
e-mail: francisco.manuel.camoes@gmail.com
tlf: (+351) 913811163

Hoping to hear back from you soon!
Best regards,
Francisco Camões

-------------------
Code:

#-- LIBRARIES --#
import pandas as pd
import numpy as np
import pandapower as pp
import random as random
import pandapower.networks as nw
import pandapower.estimation as est
import matplotlib.pyplot as plt
from pandapower.control.controller.const_control import ConstControl
from pandapower.timeseries.data_sources.frame_data import DFData
from pandapower.timeseries.run_time_series import run_timeseries
from pandapower.timeseries.output_writer import OutputWriter

#Read SE Inputs
net_1 = pp.from_pickle("net.p")
df_inputs = pd.read_pickle('measurements.pkl')
df_opf_results = pd.read_pickle('opf_results.pkl') #Para ir buscar os desvios padrão

n_variaveis = len(df_opf_results.columns)

#Variaveis Auxiliares
x = 0
count_rows = 0

# --------------------------------------------- #
# ----------------- STATE EST ----------------- #
# --------------------------------------------- #
while count_rows < 2: #len(df_inputs):

    while (x < (n_variaveis)):

        #Tensão - V, nos barramentos
        if (x < len(net_1.bus.index)):
            for bus_name in net_1.bus.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.01 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "v", "bus", df_inputs.iat[count_rows, x], desv_padrao, bus_name)
                x = x + 1
        bus_name = 0

        if ((x >= len(net_1.bus.index)) and (x < 2 * len(net_1.bus.index))):
            x = x + 1

        #Pinj, nos barramentos
        if ((x >= 2*len(net_1.bus.index)) and (x < 3*len(net_1.bus.index))):
            for bus_name in net_1.bus.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "p", "bus", df_inputs.iat[count_rows, x], desv_padrao, bus_name)
                x = x + 1
        bus_name = 0

        #Qinj, nos barramentos
        if ((x >= 3*len(net_1.bus.index)) and (x < 4*len(net_1.bus.index))):
            for bus_name in net_1.bus.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "q", "bus", df_inputs.iat[count_rows, x], desv_padrao, bus_name)
                x = x + 1
        bus_name = 0

        #Pl, nas linhas
        if ((x >= 4*len(net_1.bus.index)) and (x < (4*len(net_1.bus.index) + len(net_1.line.index)))):
            aux1 = 0
            for line_name in net_1.line.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "p", "line", df_inputs.iat[count_rows, x], desv_padrao, line_name, side='from')
                x = x + 1
        line_name = 0

        #Ql, nas linhas
        if (x >= (4*len(net_1.bus.index) + len(net_1.line.index))) and (x < (4*len(net_1.bus.index) + 2*len(net_1.line.index))):
            for line_name in net_1.line.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "q", "line", df_inputs.iat[count_rows, x], desv_padrao, line_name, side='from')
                x = x + 1

        if x >= (4*len(net_1.bus.index) + 2*len(net_1.line.index)):
            x = x + 1

    print('--------------')
    print(net_1)
    print('--------------')

    est.estimate(net_1, algorithm='wls', init="flat")
    print(net_1.res_bus_est)
    print(net_1.res_line_est)

    count_rows = count_rows + 1

    print(net_1.res_bus_est)
    print(net_1.res_line_est)

jhmenke, 2020-10-19:
Could you reformat the code properly so that we can get a better overview? Most of the code is now simply text. Thanks

fmpnsc, 2020-10-19:
@jhmenke Thank you for your attention!
Sorry for the incorrect formatting of the code, I'm pretty new to GitHub.
I'll try to paste the code in the correct format + I'll provide the .py and .pkl download link below so you can run it (and possibly debug it).
Hoping to hear from you soon, I'm in desperate need of help on this matter.

DOWNLOAD LINK:
[https://we.tl/t-zhIJ8Qx3XS](url)

```#-- LIBRARIES --#
import pandas as pd
import numpy as np
import pandapower as pp
import random as random
import pandapower.networks as nw
import pandapower.estimation as est
import matplotlib.pyplot as plt
from pandapower.control.controller.const_control import ConstControl
from pandapower.timeseries.data_sources.frame_data import DFData
from pandapower.timeseries.run_time_series import run_timeseries
from pandapower.timeseries.output_writer import OutputWriter

#Read SE Inputs
net_1 = pp.from_pickle("net.p")
df_inputs = pd.read_pickle('measurements.pkl')
df_opf_results = pd.read_pickle('opf_results.pkl') #Para ir buscar os desvios padrão

n_variaveis = len(df_opf_results.columns)
print(net_1)

#Variaveis Auxiliares
x = 0
count_rows = 0

while count_rows < 3: #len(df_inputs):
    while (x < (n_variaveis)):
        #Bus Voltages
        if (x < len(net_1.bus.index)):
            for bus_name in net_1.bus.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.01 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "v", "bus", df_inputs.iat[count_rows, x], desv_padrao, bus_name)
                x = x + 1
        bus_name = 0
        #Advance in positions of df which are not used
        if ((x >= len(net_1.bus.index)) and (x < 2 * len(net_1.bus.index))):
            x = x + 1
        #Buses Pinj
        if ((x >= 2*len(net_1.bus.index)) and (x < 3*len(net_1.bus.index))):
            for bus_name in net_1.bus.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "p", "bus", df_inputs.iat[count_rows, x], desv_padrao, bus_name)
                x = x + 1
        bus_name = 0

        #Buses Qinj
        if ((x >= 3*len(net_1.bus.index)) and (x < 4*len(net_1.bus.index))):
            for bus_name in net_1.bus.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "q", "bus", df_inputs.iat[count_rows, x], desv_padrao, bus_name)
                x = x + 1
        bus_name = 0

        #Line Active Power Flow
        if ((x >= 4*len(net_1.bus.index)) and (x < (4*len(net_1.bus.index) + len(net_1.line.index)))):
            aux1 = 0
            for line_name in net_1.line.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "p", "line", df_inputs.iat[count_rows, x], desv_padrao, line_name, side='from')
                x = x + 1
        line_name = 0

        #Line Reactive Power Flow
        if (x >= (4*len(net_1.bus.index) + len(net_1.line.index))) and (x < (4*len(net_1.bus.index) + 2*len(net_1.line.index))):
            for line_name in net_1.line.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "q", "line", df_inputs.iat[count_rows, x], desv_padrao, line_name, side='from')
                x = x + 1

        #Advance in positions of df which are not used
        if x >= (4*len(net_1.bus.index) + 2*len(net_1.line.index)):
            x = x + 1

    est.estimate(net_1, algorithm='wls', init="flat")

    print('--------------')
    print('Net Results Bus')
    print(net_1.res_bus_est)

    print('--------------')
    print('Net Results Lines')
    print(net_1.res_line_est)

    print('--------------')
    print('Contador de Linhas')
    print(count_rows)
    count_rows = count_rows + 1

print('-----------------')
print('impedance')
print(net_1.res_impedance)```

-------------------------------------

fmpnsc, 2020-10-20:
Updated my code with the help of @jhmenke.
Encountering the following errors at the moment (images presented below).
If anyone can assist me, I'd be greatly appreciated

[image]
[image]

wangzhenassd, 2020-10-20:
Can you maybe post the net with measurement as json?

fmpnsc, 2020-10-20:
@wangzhenassd 
Thanks for the attention!
I'm not sure what you mean by "posting the net with measurement as json", but I'm assuming it's for me to post the network file as .json, right?
If so, I've uploaded the net.json to this link (also upload the inputs df as json as well):
[https://we.tl/t-qUyXma2ujx](url)

wangzhenassd, 2020-10-20:
Your code actually works if I simply comment this line out. I think you should check the logic of deleting measurements.
```python
#Limpar Measurements
    net_1.measurement.drop(net_1.measurement.index, inplace=True)
```

wangzhenassd, 2020-10-20:
```python
import pandas as pd
import numpy as np
import pandapower as pp
import random as random
import pandapower.networks as nw
import pandapower.estimation as est
import matplotlib.pyplot as plt
from pandapower.control.controller.const_control import ConstControl
from pandapower.timeseries.data_sources.frame_data import DFData
from pandapower.timeseries.run_time_series import run_timeseries
from pandapower.timeseries.output_writer import OutputWriter

# #Read SE Inputs
# net_1 = pp.from_pickle("net.p")
# df_inputs = pd.read_pickle('measurements.pkl')
# df_opf_results = pd.read_pickle('opf_results.pkl') #Para ir buscar os desvios padrão

df_inputs = pd.read_json("measurements_json.json")
df_opf_results = df_inputs
net_1 = pp.from_json("net_json.json")

n_variaveis = len(df_opf_results.columns)
print(net_1)

#Variaveis Auxiliares
x = 0
count_rows = 0

while count_rows < 3: #len(df_inputs):

   #Limpar Measurements
    # net_1.measurement.drop(net_1.measurement.index, inplace=True)

    while (x < (n_variaveis)):
        #Bus Voltages
        if (x < len(net_1.bus.index)):
            for bus_name in net_1.bus.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.01 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "v", "bus", df_inputs.iat[count_rows, x], desv_padrao, bus_name)
                x = x + 1
        bus_name = 0
        #Advance in positions of df which are not used
        if ((x >= len(net_1.bus.index)) and (x < 2 * len(net_1.bus.index))):
            x = x + 1
        #Buses Pinj
        if ((x >= 2*len(net_1.bus.index)) and (x < 3*len(net_1.bus.index))):
            for bus_name in net_1.bus.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "p", "bus", df_inputs.iat[count_rows, x], desv_padrao, bus_name)
                x = x + 1
        bus_name = 0

        #Buses Qinj
        if ((x >= 3*len(net_1.bus.index)) and (x < 4*len(net_1.bus.index))):
            for bus_name in net_1.bus.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "q", "bus", df_inputs.iat[count_rows, x], desv_padrao, bus_name)
                x = x + 1
        bus_name = 0

        #Line Active Power Flow
        if ((x >= 4*len(net_1.bus.index)) and (x < (4*len(net_1.bus.index) + len(net_1.line.index)))):
            aux1 = 0
            for line_name in net_1.line.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "p", "line", df_inputs.iat[count_rows, x], desv_padrao, line_name, side='from')
                x = x + 1
        line_name = 0

        #Line Reactive Power Flow
        if (x >= (4*len(net_1.bus.index) + len(net_1.line.index))) and (x < (4*len(net_1.bus.index) + 2*len(net_1.line.index))):
            for line_name in net_1.line.index:
                desv_padrao = abs(df_opf_results.iat[count_rows, x]) * 0.02 / 3
                if (desv_padrao == 0):
                    desv_padrao = 0.000001
                print(desv_padrao)
                pp.create_measurement(net_1, "q", "line", df_inputs.iat[count_rows, x], desv_padrao, line_name, side='from')
                x = x + 1

        #Advance in positions of df which are not used
        if x >= (4*len(net_1.bus.index) + 2*len(net_1.line.index)):
            x = x + 1

    print(net_1.measurement.to_string())
    est.estimate(net_1, algorithm='wls', init="flat")

    print('--------------')
    print('Net Results Bus')
    print(net_1.res_bus_est)

    print('--------------')
    print('Net Results Lines')
    print(net_1.res_line_est)

    print('--------------')
    print('Contador de Linhas')
    print(count_rows)
    count_rows = count_rows + 1
```

fmpnsc, 2020-10-21:
@wangzhenassd Thank you for your answer.
Didn't you get the "Estimation failed! Pandapower network failed to update!" Error?
I copied the code you put here and it still gets me this same error.

wangzhenassd, 2020-10-21:
Can you try update your pandapower version?

fmpnsc, 2020-10-21:
@wangzhenassd 
Thank you so much for all the insightful tips.
The script is finally working. Don't know why, but updating from 2.3.1 to 2.4 completely fixed this issue.

wangzhenassd, 2020-10-22:
The reason is that since pandapower 2.4.0, the pq measurement of bus changed its sign system to match the rest of the pandapower convention. If you are using pandapower < 2.4.0, you need to revert the sign of bus pq measurment. Please reopen if there are further issues.

closed 2020-10-22 by wangzhenassd