e2nIEE/pandapower #956: Hello,I have a bug , when I watch your youtube

learningelectric, 2020-10-20:
this is code in pycharm:
import pandas as pd
import pandapower as pp

net = pp.create_empty_network()
df = pd.read_excel("grid_data.xlsx", sheet_name="bus", index_col=0)
#print(df)
for idx in df.index:
    pp.create_bus(net, vn_kv = df.at[idx, "vn_kv"])

print(net)

this is output:

Traceback (most recent call last):
  File "D:\anaconda3\envs\pandapower\lib\site-packages\pandas\core\indexes\base.py", line 2890, in get_loc
    return self._engine.get_loc(key)
  File "pandas\_libs\index.pyx", line 107, in pandas._libs.index.IndexEngine.get_loc
  File "pandas\_libs\index.pyx", line 131, in pandas._libs.index.IndexEngine.get_loc
  File "pandas\_libs\hashtable_class_helper.pxi", line 1607, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas\_libs\hashtable_class_helper.pxi", line 1614, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'vn_kv'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:/pycharm/pandapower/readdata2020-10-20/read_data.py", line 8, in <module>
    pp.create_bus(net, vn_kv = df.at[idx, "vn_kv"])
  File "D:\anaconda3\envs\pandapower\lib\site-packages\pandas\core\indexing.py", line 2175, in __getitem__
    return self.obj._get_value(*key, takeable=self._takeable)
  File "D:\anaconda3\envs\pandapower\lib\site-packages\pandas\core\frame.py", line 2823, in _get_value
    series = self._get_item_cache(col)
  File "D:\anaconda3\envs\pandapower\lib\site-packages\pandas\core\generic.py", line 3271, in _get_item_cache
    values = self._data.get(item)
  File "D:\anaconda3\envs\pandapower\lib\site-packages\pandas\core\internals\managers.py", line 948, in get
    loc = self.items.get_loc(item)
  File "D:\anaconda3\envs\pandapower\lib\site-packages\pandas\core\indexes\base.py", line 2892, in get_loc
    return self._engine.get_loc(self._maybe_cast_indexer(key))
  File "pandas\_libs\index.pyx", line 107, in pandas._libs.index.IndexEngine.get_loc
  File "pandas\_libs\index.pyx", line 131, in pandas._libs.index.IndexEngine.get_loc
  File "pandas\_libs\hashtable_class_helper.pxi", line 1607, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas\_libs\hashtable_class_helper.pxi", line 1614, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'vn_kv'

I really don't know why the bug create. And I want to know how to solve it. 
please help me or give me some suggestions.
Thank you very much!
(PS:I am not a native English speaker. Please forgive me if I have been rude in some places)

rbolgaryn, 2020-10-20:
Hi @learningelectric ,

seems like there is an issue with df?
Can you check if the df has been read correctly and has the columns that you need, e.g. vn_kv?

Roman

learningelectric, 2020-10-21:
In your advice, I check my df and then I find my low-level error.I really feel shame because of the low-level error that I made by accident. I am very sorry and I am really  grateful for your generous and zealous help for me.
At last, I am honor that you can work successfully and own a healthy body. Thank you very much!

rbolgaryn, 2020-10-21:
You are welcome!

closed 2020-10-21 by rbolgaryn