e2nIEE/pandapower #959: substation - what is their definition? - I cannot find the data for it

Marahop, 2020-10-20:
Hi everyone! 

I would like to know the definition of the substation used in pandapower. How many transformers are considered to be in one substation? Are the transformers in a substation also listed in the transformer list? 
Where can I find its definition/code?

Thank you very much!

lthurner, 2020-10-20:
As far as I know, there is no such thing as a substation in pandapower. Where did you come across substations?

Marahop, 2020-10-20:
When I call all the elements of the pandapower network Simbench, I get the following list: 
"this pandapower network includes the following parameter tables:
- bus (100 elements)
- load (98 elements)#.
...
- trafo (2 elements)
...
- substation (2 elements)
..."

the substations names I can also extract, they are HV1_MV4.101_Substation (voltLvl 4) & MV4.101_Remote_Station (volLvl 5)

I cannot find anything about substations in the documentation online, they are mentioned in the Simbench pdf though, but only mentioned - no definition or explanation.

Any help very appreciated :) Thank you!
An help

lthurner, 2020-10-20:
The substation table in simbench was added since the substation information is included in the simbench data. It has no function or influence for pandapower. If there is any documentation about theses substations, you would have to look for it in the simbench docs.

closed 2020-10-20 by lthurner