e2nIEE/pandapower #961: Reference to net-object in controllers breaks when copying the net

TOffergeld, 2020-10-21:
# Description
When a `pandapowerNet` is copied, the references in its controllers (`controller.net`) break (`controller.net` is `None`), causing timeseries-simulations (`run_timeseries`) to fail.

Obviously the backref cannot be updated from within `__init__` (where the copy is created) as the correct object does not exist yet. Maybe this method of copying the `pandapowerNet` should be deprecated in favor of a method that copies the net and automatically fixes the references?

Might be related to #916 

## Steps to reproduce
1. Create a network with a controller (e.g. `ConstControl`)
2. Create a copy of the net (using `copied = pandapowerNet(original)`)
3. Verify that `original.controller.at[0, "object"].net is net`
4. Note that `copied.controller.at[0, "object"].net is None` (should be `copied`)

rbolgaryn, 2020-10-22:
Hi @TOffergeld ,

1) there is a method net.deepcopy() which returns a copy of the net, also copy.deepcopy(net) leads to the same result bacause it uses net.deepcopy().
You can also take a look at the tests that cover this behavior.

2) in the latest version of pandapower (I think at the moment it's only on the develop branch) we have removed all references to net within the controllers to avoid circular references (as described in #916). This change will be released for the new version of pandapower (if not yet happened). In the meantime, you can check out the develop branch if you want to start using the latest implementation of controllers.

Roman

lthurner, 2020-10-22:
> at the moment (it's only on a separate branch clean_control) we have removed all references to net within the controllers to avoid circular references (as described in #916). 

It is already on develop, but not yet released.

rbolgaryn, 2020-11-06:
No new inputs, therefore closing for now. @TOffergeld feel free to reopen.

closed 2020-11-06 by rbolgaryn