e2nIEE/pandapower #962: et base values

MMajidi137, 2020-10-22:
Hi all!
while creating an empty network by pandapower.create_empty_network(name='', f_hz=50.0, sn_mva=1, add_stdtypes=True) there is an option to set s_base.

However, when I call a network from pandapower.network, simbench, or converting net from other formats, there is no option to change the s_base. Also, I could not find a function in pandapower.toolbox to change base values. I change it manually in net["sn_mva"]? 

I wonder to know if there is another way to set base values. Is there any callable function?

rbolgaryn, 2020-10-22:
Hi Maryam,

you can just set it manually, the way you mentioned. The values of impedance and power are not in per unit system, therefore you don't need to adjust anything else apart from sn_mva

MMajidi137, 2020-10-22:
Thanks, @rbolgaryn. so when I need to check other base values (I, v, z) I should calculate by myself. am I right?

rbolgaryn, 2020-10-22:
How do you check the base values otherwise?

MMajidi137, 2020-10-22:
> How do you check the base values otherwise?

I calculate them. I was just curious to know if there is any available function in pandapower to calculate these values.

rbolgaryn, 2020-10-22:
I think the base values are calculated "under the hood" during load flow. Maybe it would be useful to add such a function to pp.toolbox, which would convert the element parameters to/from per-unit system.

rbolgaryn, 2020-10-22:
Since you already are calculating it "by hand", would you like to implement it for pp.toolbox?

MMajidi137, 2020-10-22:
great! vielen Dank

MMajidi137, 2020-10-22:
> Since you already are calculating it "by hand", would you like to implement it for pp.toolbox?

yes. sure

SteffenMeinecke, 2021-11-11:
@MMajidi137 is this issue already fixed?

MMajidi137, 2021-11-11:
> @MMajidi137 is this issue already fixed?
you may close it

MMajidi137, 2021-11-11:
@SteffenMeinecke you may close this issue

closed 2021-11-11 by SteffenMeinecke