e2nIEE/pandapower #963: hello! maybe this is a bug about "Example simple plot on a map" in your document

learningelectric, 2020-10-22:
this is document's website:   

https://pandapower.readthedocs.io/en/v2.4.0/plotting/plotly/built-in_plots.html

In this document, when you use code in pycharm:
"Example power flow results plot" :
..............................................................................................................................................
from pandapower.plotting.plotly import simple_plotly
from pandapower.networks import mv_oberrhein
net = mv_oberrhein()
pf_res_plotly(net, on_map=True, projection='epsg:31467', map_style='dark')
...............................................................................................................................................

when you use this code, you will have a bug and then you only can see white website. you can not see network in website. 
Actually, I really don't know how to solve this issue or error as a rookie who is interested in pandapower.
So I guess that in this document some codes has been outdate.

I clearlly know you are very busy as an engineer ,but in fact I think pandapower's plot is very beautiful.
It is very cool!!  It should be known by more good engineers. But it is very  embarrassing that in the latest version many beautiful plots have some bugs specially like "pf_res_plotly(net, on_map=True, projection='epsg:31467', map_style='dark')" and "vlevel_plotly()".

Maybe this is not a bug because of my mistakes , but I really hope you can make more tutorials by your youtube  about how to make plots using geodate network in your spare time. Or you can update some new codes to replace some old code in your website.  If you can help some rookies like me, I really appreciate your generous help very much. I really don't know whether it is a rude request. But as you know , it is really difficult for a rookie and a developer to find these bugs and solve these bugs.

At last , you do not need make some tutorials or update some new codes as soon as possible, because I know you have some other main porjects to need you finish. I sincerely wish you success in your work and good health and then give us some helps. 

Thank you very much!

(PS: because I am not a native speaker of English ,my some sentences are translated by translation software. If there are some rude sentences in my words, I sincerely beg you can pardon me !)

learningelectric, 2020-10-22:
this is output in pycharm:

Test if geo-data are in lat/long cannot be performed using geopy -> eventual plot errors are possible.

rbolgaryn, 2020-11-06:
Hi @SteffenMeinecke , @SimonRubenDrauz ,

I can see that there is some code from you in the function, so you probably know how it works.
Can you please take a look?

Roman

jkisse, 2020-11-11:
@learningelectric Have you tried this tutorial: https://github.com/e2nIEE/pandapower/blob/master/tutorials/plotly_maps.ipynb ? 
Do you have a mapbox-Account and have you set your mapbox-token?

rbolgaryn, 2020-12-03:
no updates so far, closing for now.

closed 2020-12-03 by rbolgaryn