e2nIEE/pandapower #965: Creating Transformer - how to consider ohmic losses?

Marahop, 2020-10-26:
Hi everyone!

I have watched almost all pandapower introduction videos now and went throught the documentation (https://pandapower.readthedocs.io/en/v2.4.0/elements/trafo.html#pandapower.create_transformer_from_parameters). 
I still have a few questions regarding the definition/creation of a transformer.

What interest me are the **losses**.
In the documentation it mentions two types of losses, right?!:

pfe_kw* - iron losses
i0_percent* - open loop losses in [%]

So I wonder where the **ohmic/copper/core losses** are being considered? 
They aren't considered?  Neither in _pandapower.create_transformer(net, hv_bus, lv_bus, std_type, name=None, tap_pos=nan, in_service=True, index=None, max_loading_percent=nan, parallel=1, df=1.0_
nor in 
_pandapower.create_transformer_from_parameters(net, hv_bus, lv_bus, sn_mva, vn_hv_kv, vn_lv_kv, vkr_percent, vk_percent, pfe_kw, i0_percent, shift_degree=0, tap_side=None, tap_neutral=nan, tap_max=nan, tap_min=nan, tap_step_percent=nan, tap_step_degree=nan, tap_pos=nan, tap_phase_shifter=False, in_service=True, name=None, vector_group=None, index=None, max_loading_percent=nan, parallel=1, df=1.0, vk0_percent=nan, vkr0_percent=nan, mag0_percent=nan, mag0_rx=nan, si0_hv_partial=nan, **kwargs)_ ?

I would be very happy for your help. 
Thank you very much!

rbolgaryn, 2020-10-28:
Dear @Marahop ,

transformers also have parameters short-circuit voltage vk_percent and the real part of the short-circuit voltage vkr_percent, which are evaluated in a transformer test in practice (Short-circuit test, Kurzschlussversuch). You should be able to calculate the ohmic losses based on these parameters, as commonly described in literature.

lthurner, 2020-10-29:
Closed - please reopen if you have further questions.

closed 2020-10-29 by lthurner