e2nIEE/pandapower #966: Admittance Matrix after Bus-Element Switching off ...

Houshyar20, 2020-10-26:
Hi everybody,

I have two questions which will come below. But let me explain a little bit about the story of my problem.
By use of Pandapower, thanks to the developers of Pandapower, I want to calculate current deviations in different lines of a network when an arbitrary line of network is removed. I consider a small network such as CIGRE. First, by considering all switches are closed, I run power flow and extract the observational voltages and currents (let's denote them by I_bf = net.res_line.i_ka and E_bf = Vm*(cos(Va) + complex(0,1)*sin(Va)) with Vm = net.res_bus_vm_pu, Va = net.res_bus.va_degree*pi/180).
In order to remove a line, e.g. the line 12 (connected to buses 7 and 6), I open the switch S2 with  index of 1 as net.switch.closed [1] = False.
And again, I run power flow and extract the voltages and currents (let's denote them by I_af = net.res_line.i_ka and E_af = Vm*(cos(Va_af) + complex(0,1)*sin(Va_af)), where Vm(Va) are obtained as above).

Now, I calculate the deviations of observed voltages and currents as follows:
DE = E_af - E_bf
DI_ab = abs(D_E[a] - D_E[b]/z_ab)
DI_o = I_af - I_bf
Relative Error = abs(DI_o - DI_ab)*100/DI_o,
where the impedance of line ab is z_ab = r_ab + i*x_ab with r_ab (x_ab) as the resistance (reactance) of the line connected between buses a and b. z_ab is also extracted by Pandapower.
Now, I have two questions:
1- I expect that the observed current deviations (DI_o) and the calculated ones (DI_ab) have similar values (for example, with less than 10% relative errors). But I obtain for 7 lines out of 15 total lines 
an error bigger than 10% within which I even have errors more than 100%!! 
For me, it is not reasonable. I'm not sure, is there any problem in Pandapower data which are my input?
I'm using the standard formula in my calculations and the code is so simple 
( [D_VI.txt](https://github.com/e2nIEE/pandapower/files/5440931/D_VI.txt) ) .

2- My second question is related to lack of the enough information about Pypower/Pandapower. In Pandapower, by removing a line and running power flow, we obtain an admittance matrix with higher dimension (e.g., for CIGRE, before removing a line: Ybus is a 14*14 matrix but after removing one line(or two lines), Ybus is a 15*15 (or 16*16) matrix. Why should Pandapower make extra dimension(s) for Ybus matrix?
And I'd like to know how we can find the position of extra row(s) and column(s) in the new Ybus (I mean Ybus after removing line(s) )?

I hope you, especially the developers, have time to reply/leave comments to resolve aformentioned problem and questions.

Kind regards,

lthurner, 2020-10-28:
Why your calculation is off I can't say, you will have to check in detail where you went wrong.

The differences in Ybus dimension are due to the auxiliary buses that are created for open switches. This is explained in the documentation: https://pandapower.readthedocs.io/en/v2.4.0/elements/switch.html#electric-model

Houshyar20, 2020-10-29:
@lthurner 
Thank you Leon for your reply and comments.
- I didn't understand your mean. Which part of my calculation seems to be off. Maybe, it seems unclear because of my notation ... I used just standard formula to find deviations of voltage and current. To calculate the current in line "ab" (connecting bus "a" and "b"), I used the standard formula: I_ab = (E_a - E_b)/z_ab with E as voltage and z_ab as the impedance of the line "ab". I also attached my simple code.
So, is there any unclear or wrong point in using these standard formulas?
On the other hand, I have a simple code to extract voltage and current from pandapower. I think, everything seems to be OK in extracting voltage and current.
As a result, I think, the only part for which we can discuss is the voltage and current values from pandapower. I don't know what's your mean ...

- Thanks for sending the link. I had read the document before. But it's still unclear for me how I can recognize the extra row and column in the new Ybus matrix (Ybus after removing one line of the network) in pandapower. 
Thanks to pandapower by keeping the same index for auxiliary bus unlike pypower and that is a helpeful point. But in the calculation of current by use of Ybus, after removing a few lines, I get a problem because of the dimension of the new Ybus (let me denote it by Ynew):
I_ab = Ynew_ab.(V_a - V_b).
For example for CIGRE, after removing two lines, I want to calculate the absolute value of I_ab vector and compare with the observational current vector in pandapower (that is, net.res_line.i_ka). The observational current vector has 15 elements whereas in the calculation of current I_ab, our Ynew is an 18x18 matrix!
That's a problem for me to calculate I_ab by using this standard formula and compare with the observational current from pandapower (net.res_line.i_ka).

Thank you to spending time to resolve the problem which I think, if I'm not wrong, it may be from the pandapower data. 

Best regards,

rbolgaryn, 2020-10-29:
Hi @Houshyar20 ,

as for the first point, I am sorry that we cannot invest the time to go through the specific use-case and debug it. You would need to debug the code and the calculation on your own. In case you identify that the error is caused by a mistake in pandapower, please let us know.

As for the second point, maybe it will be easier for you to set the line out of service instead of opening the switch. In that case, there should not be any additional buses in the Y matrix (it would then be smaller than the matrix for the grid with all lines in service). The charging currents for the open line would not be modelled properly, but I assume that the impact on the results is small enough to omit it. 

Alternatively, you can add auxiliary buses yourself and replace the bus-line switches by bus-bus switches. In that case, if you open a switch, the matrix should have the same dimensions, as the number of buses remains the same.

Roman

Houshyar20, 2020-10-29:
@lthurner 

OK. Let me please debug it and focus more on the problem. I will come back to you if I found any problem in pandapower data.

Many thanks to you for your informative comments about fixing the Ybus matrix extra dimensions. 

Kind regards,

lthurner, 2020-10-29:
About the Ybus dimensions, you can also check out this tutorial: https://github.com/e2nIEE/pandapower/blob/develop/tutorials/internal_datastructure.ipynb

closed 2020-10-29 by lthurner