e2nIEE/pandapower #968: Example of unbalanced network

AdrienGougeon, 2020-10-28:
Hello,
I am trying to build an unbalanced power network but I have a lot of warning due to the usage of an impedance when I run the unbalanced powerflow : https://framabin.org/p/?0940223500f6c431#6CqBHNxIUCnhdnxIIFuq8qXmfTxSd6PCnLQrgkUEmRs=

Without the impedance the unbalanced powerflow works fine, it also works fine if I use the impedance and run a balanced powerflow.

Is there any example of a clean unbalanced network using an impedance to help me find where is my mistake?

lthurner, 2020-10-28:
Yes your are right, the impedance element is not yet implemented in the unbalanced power flow. It only has R/X values, and no parameters for zero sequence. What would be the desired behaviour here in your opinion?

rbolgaryn, 2020-11-06:
Has not been updated in a while, closing for now. @AdrienGougeon feel free to reopen if you have additional information.

AnBlav, 2020-11-18:
Dear @rbolgaryn,

Thanks for your answer. Hence, for running an unbalanced power flow with a non-zero impedance representing the "strength" of the network (in order to have non-ideal voltage source), do I need to replace the impedance element (not available for unbalanced power flows) with another element, such as a line element in which negative-positive and zero sequences parameters are accessible?

Thanks in advance for your answer.

Best regards,

Anne

AnBlav, 2020-12-04:
Dear @lthurner,

Thanks for your answer. Hence, for running an unbalanced power flow with a non-zero impedance representing the "strength" of the network (in order to have non-ideal voltage source), do I need to replace the impedance element (not available for unbalanced power flows) with another element, such as a line element in which negative-positive and zero sequences parameters are accessible?

Thanks in advance for your answer.

Best regards,

Anne

lthurner, 2020-12-08:
The ext_grid element has parameters for internal short-circuit parameters: s_sc_max_mva, rx_max, r0x0_max, x0x_max. These are meant to represent the internal resistance of the network. Can't you use those parametesr?

AnBlav, 2020-12-08:
Dear @lthurner ,

Thanks for your answer: we hadn't seen that this element could be used for unbalanced power flow. That solves our problem.

Best regards,

Anne

closed 2020-12-08 by lthurner