e2nIEE/pandapower #971: TIme series example from youtube

bmitrauncc, 2020-10-29:
I was trying to replicate the time series example shown in youtube "https://www.youtube.com/watch?v=sAHoJbfLhas".
But I keep getting a flat line in the second plot.

Also do you know how were the p.u values obtained for pv, wind and residential?

If I try to plot the net shown in the example using pandapower.plotting.simple_plot(net), I get an error saying pandapower has no attribute plotting.

rbolgaryn, 2020-11-06:
You should first import it manually:

import pandapower.plotting

After that, you should be able to use it the way you tried in your comment.

rbolgaryn, 2020-11-06:
Also, make sure that you have uigraph installed (conda install python-igraph)

closed 2020-11-06 by rbolgaryn