e2nIEE/pandapower #972: Plotly issue on latest pandapower version

upslowlander, 2020-10-30:
Hi! I am trying to plot a network topology using plotly (simple_plotly(net)) and I get this error:

simple_plotly(net)
Traceback (most recent call last):

  File "<ipython-input-13-c4a13aac397f>", line 1, in <module>
    simple_plotly(net)

  File "C:\Users\upslo\anaconda3\lib\site-packages\pandapower\plotting\plotly\simple_plotly.py", line 171, in simple_plotly
    return draw_traces(line_traces + trafo_trace + ext_grid_trace + bus_trace,

  File "C:\Users\upslo\anaconda3\lib\site-packages\pandapower\plotting\plotly\traces.py", line 688, in draw_traces
    fig['layout']['width'], fig['layout']['height'] = ([ar * figsize * 700 for ar in aspectratio])

  File "C:\Users\upslo\anaconda3\lib\site-packages\plotly\basedatatypes.py", line 4296, in __setitem__
    super(BaseLayoutHierarchyType, self).__setitem__(prop, value)

  File "C:\Users\upslo\anaconda3\lib\site-packages\plotly\basedatatypes.py", line 3353, in __setitem__
    self._set_prop(prop, value)

  File "C:\Users\upslo\anaconda3\lib\site-packages\plotly\basedatatypes.py", line 3636, in _set_prop
    raise err

  File "C:\Users\upslo\anaconda3\lib\site-packages\plotly\basedatatypes.py", line 3631, in _set_prop
    val = validator.validate_coerce(val)

  File "C:\Users\upslo\anaconda3\lib\site-packages\_plotly_utils\basevalidators.py", line 787, in validate_coerce
    self.raise_invalid_val(v)

  File "C:\Users\upslo\anaconda3\lib\site-packages\_plotly_utils\basevalidators.py", line 273, in raise_invalid_val
    raise ValueError(

ValueError: 
    Invalid value of type 'numpy.float64' received for the 'width' property of layout
        Received value: 2.5925925925925926

    The 'width' property is a number and may be specified as:
      - An int or float in the interval [10, inf]

I used the same code before and had no problems. Had to reinstall my system and I reinstalled anaconda, pandapower and the dependancies and get this error now. Is this a new version problem? Maybe a compatibility problem. All libraries were installed today so I have the latest versions.

Thank you.

rbolgaryn, 2020-11-06:
Can you provide a concise test/example?

jkisse, 2020-11-11:
Which Version of pandapower are you using and what is your Python version?

jkisse, 2020-11-11:
It works for me in this environment...
```
# Name                    Version                   Build  Channel
atomicwrites              1.4.0                      py_0
attrs                     20.3.0             pyhd3eb1b0_0
ca-certificates           2020.10.14                    0
certifi                   2020.6.20          pyhd3eb1b0_3
cffi                      1.14.3                   pypi_0    pypi
colorama                  0.4.4                      py_0
cryptography              3.2.1                    pypi_0    pypi
cycler                    0.10.0                   pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
iniconfig                 1.1.1                      py_0
kiwisolver                1.3.1                    pypi_0    pypi
matplotlib                3.3.2                    pypi_0    pypi
more-itertools            8.5.0                      py_0
networkx                  2.5                      pypi_0    pypi
numpy                     1.19.4                   pypi_0    pypi
openssl                   1.1.1h               he774522_0
packaging                 20.4                       py_0
pandapower                2.4.0                    pypi_0    pypi
pandas                    1.1.4                    pypi_0    pypi
pillow                    8.0.1                    pypi_0    pypi
pip                       20.2.4           py38haa95532_0
plotly                    4.12.0                   pypi_0    pypi
pluggy                    0.13.1                   py38_0
py                        1.9.0                      py_0
pycparser                 2.20                     pypi_0    pypi
pyparsing                 2.4.7                      py_0
pytest                    6.1.1                    py38_0
python                    3.8.0                hff0d562_2
python-dateutil           2.8.1                    pypi_0    pypi
python-igraph             0.8.3                    pypi_0    pypi
pytz                      2020.4                   pypi_0    pypi
retrying                  1.3.3                    pypi_0    pypi
scipy                     1.5.4                    pypi_0    pypi
setuptools                50.3.1           py38haa95532_1
six                       1.15.0                     py_0
sqlite                    3.33.0               h2a8f88b_0
texttable                 1.6.3                    pypi_0    pypi
toml                      0.10.1                     py_0
vc                        14.1                 h0510ff6_4
vs2015_runtime            14.16.27012          hf0eaf9b_3
wheel                     0.35.1                     py_0
wincertstore              0.2                      py38_0
xlrd                      1.2.0                    pypi_0    pypi
xlsxwriter                1.3.7                    pypi_0    pypi
zlib                      1.2.11               h62dcd97_4
```
```
import pandapower.networks as nw
net = nw.mv_oberrhein()
from pandapower.plotting.plotly.simple_plotly import simple_plotly
simple_plotly(net)
```

upslowlander, 2020-11-11:
> It works for me in this environment...
>
> ```
> # Name                    Version                   Build  Channel
> atomicwrites              1.4.0                      py_0
> attrs [...]

Hi jkisse

This is the environment:

# Name                    Version                   Build  Channel
_anaconda_depends         2020.07                  py38_0  
_ipyw_jlab_nb_ext_conf    0.1.0                    py38_0  
_libgcc_mutex             0.1                        main  
alabaster                 0.7.12                     py_0  
anaconda                  custom                   py38_1  
anaconda-client           1.7.2                    py38_0  
anaconda-navigator        1.10.0                   py38_0  
anaconda-project          0.8.4                      py_0  
argh                      0.26.2                   py38_0  
argon2-cffi               20.1.0           py38h7b6447c_1  
asn1crypto                1.4.0                      py_0  
astroid                   2.4.2                    py38_0  
astropy                   4.0.2            py38h7b6447c_0  
async_generator           1.10                       py_0  
atomicwrites              1.4.0                      py_0  
attrs                     20.2.0                     py_0  
autopep8                  1.5.4                      py_0  
babel                     2.8.0                      py_0  
backcall                  0.2.0                      py_0  
backports                 1.0                        py_2  
backports.functools_lru_cache 1.6.1                      py_0  
backports.shutil_get_terminal_size 1.0.0                    py38_2  
backports.tempfile        1.0                        py_1  
backports.weakref         1.0.post1                  py_1  
beautifulsoup4            4.9.3              pyhb0f4dca_0  
bitarray                  1.6.0            py38h7b6447c_0  
bkcharts                  0.2                      py38_0  
blas                      1.0                         mkl  
bleach                    3.2.1                      py_0  
blosc                     1.20.1               hd408876_0  
bokeh                     2.2.3                    py38_0  
boto                      2.49.0                   py38_0  
bottleneck                1.3.2            py38heb32a55_1  
brotlipy                  0.7.0           py38h7b6447c_1000  
bzip2                     1.0.8                h7b6447c_0  
ca-certificates           2020.10.14                    0  
cairo                     1.14.12              h8948797_3  
certifi                   2020.6.20          pyhd3eb1b0_3  
cffi                      1.14.3           py38he30daa8_0  
chardet                   3.0.4                 py38_1003  
click                     7.1.2                      py_0  
cloudpickle               1.6.0                      py_0  
clyent                    1.2.2                    py38_1  
colorama                  0.4.4                      py_0  
conda                     4.9.1            py38h06a4308_0  
conda-build               3.20.5                   py38_1  
conda-env                 2.6.0                         1  
conda-package-handling    1.7.2            py38h03888b9_0  
conda-verify              3.4.2                      py_1  
contextlib2               0.6.0.post1                py_0  
cryptography              3.1.1            py38h1ba5d50_0  
curl                      7.71.1               hbc83047_1  
cycler                    0.10.0                   py38_0  
cython                    0.29.21          py38he6710b0_0  
cytoolz                   0.11.0           py38h7b6447c_0  
dask                      2.30.0                     py_0  
dask-core                 2.30.0                     py_0  
dbus                      1.13.18              hb2f20db_0  
[... 250 lines cut ...]
wcwidth                   0.2.5                      py_0  
webencodings              0.5.1                    py38_1  
werkzeug                  1.0.1                      py_0  
wheel                     0.35.1                     py_0  
widgetsnbextension        3.5.1                    py38_0  
wrapt                     1.11.2           py38h7b6447c_0  
wurlitzer                 2.0.1                    py38_0  
xlrd                      1.2.0                      py_0  
xlsxwriter                1.3.7                      py_0  
xlwt                      1.3.0                    py38_0  
xmltodict                 0.12.0                     py_0  
xz                        5.2.5                h7b6447c_0  
yaml                      0.2.5                h7b6447c_0  
yapf                      0.30.0                     py_0  
zeromq                    4.3.3                he6710b0_3  
zict                      2.0.0                      py_0  
zipp                      3.4.0              pyhd3eb1b0_0  
zlib                      1.2.11               h7b6447c_3  
zope                      1.0                      py38_1  
zope.event                4.5.0                    py38_0  
zope.interface            5.1.2            py38h7b6447c_0  
zstd                      1.4.5                h9ceee32_0

Thank you
Best Regards

jkisse, 2020-11-11:
Thank you @upslowlander. 
Does this work for you:
```
import pandapower.networks as nw
net = nw.mv_oberrhein()
from pandapower.plotting.plotly.simple_plotly import simple_plotly
simple_plotly(net)
```

Please provide a minimal example to reproduce your issue.

rbolgaryn, 2020-12-02:
Hi @upslowlander , I cannot reproduce the issue, and we haven't received a responce in a while. Feel free to reopen the issue if you have any updates.

Roman

closed 2020-12-02 by rbolgaryn