e2nIEE/pandapower #973: Pandapower Voltagedrop and trafo control

mohammadswaiti, 2020-11-01:
i am trying to get the voltage drop problem out by controlling the transformer in the following net 380 kv bus connected to Transformer than the transformer is connected to low voltage bus with 110 kv and this bus connected to a 50 km line that connected to the third bus 'b3' and there is a load connected to the 'b3' bus , the question is how can i control the transformer 'Trafo 1' to get the voltage on the bus 'b3' 110kv and avoid the voltage drop

with my best regards

`import pandapower as pp 
from pandapower.control.controller.trafo_control import TrafoController
net = pp.create_empty_network() #create an empty network
b1 = pp.create_bus(net,vn_kv=380,type='b',name='bus1',in_service = True)
b2 = pp.create_bus(net,vn_kv=110,type='b',name='bus2',in_service = True)
b3 = pp.create_bus(net,vn_kv=110,type='b',name='bus3',in_service = True)
trafo1 = pp.create_transformer(net, b1, b2, name="trafo1" ,std_type="160 MVA 380/110 kV",in_service = True)
l1 = pp.create_line(net,b2,b3,name = '110kV line',length_km = 50.0,std_type = '305-AL1/39-ST1A 110.0', max_loading_percent=735.0,in_service = True)
load = pp.create_load(net,b3,p_mw=72,q_mvar=34.87, in_service=True)
ext = pp.create_ext_grid(net, b1,voltage = 1.03,name = 'extra_grid',in_service = True `

rbolgaryn, 2020-11-02:
Dear @mohammadswaiti ,

you can implement your own transformer Controller, e.g. by inheritance from the class TrafoController or DiscreteTapControl. 
You can refer to the tutorials in pandapower regarding controller elements. You could also try setting the attribute of DiscreteTapControl to the remote bus and see if it works that way. If you implement the transformer tap changer for remote buses, feel free to open a pull request!

Roman

mohammadswaiti, 2020-11-02:
thank you a lot for your answer i have used the trafo controller and it works 
i will add the code that i used to solve this problem 

`import pandapower as pp 
import pandapower.control as control
net = pp.create_empty_network() #create an empty network
b1 = pp.create_bus(net,vn_kv=380,type='b',name='bus1',in_service = True)
b2 = pp.create_bus(net,vn_kv=110,type='b',name='bus2',in_service = True)
b3 = pp.create_bus(net,vn_kv=110,type='b',name='bus3',in_service = True)

trafo1 = pp.create_transformer(net, b1, b2, name="trafo1" ,std_type="160 MVA 380/110 kV",in_service = True)
#trafo_controller = control.DiscreteTapControl(net=net, tid=0,vm_upper_pu= 1.0, vm_lower_pu=1.13, vm_uppe_pur=1.01)
trafo_controller = control.ContinuousTapControl(net=net, tid=0,vm_upper_pu= 1.12,vm_set_pu = 1.08, vm_uppe_pur=1)
l1 = pp.create_line(net,b2,b3,name = '110kV line',length_km = 50.0,std_type = '305-AL1/39-ST1A 110.0', max_loading_percent=735.0,in_service = True)
load = pp.create_load(net,b3,p_mw=72,q_mvar=34.87, in_service=True)
ext = pp.create_ext_grid(net, b1,voltage = 1.03,name = 'extra_grid',in_service = True)`

rbolgaryn, 2020-11-03:
Hi @mohammadswaiti ,

this controls the voltage at bus b2 then, and not b3. 
Have you found a way to control the voltage at bus b3?

Roman

mohammadswaiti, 2020-11-05:
hi Roman , 
i  want to find a way to control the voltage on  bus b3 using the Trafo Control   but i found noway yet,
i will try to use the DiscreteTapControl , if i found any thing i will leave it as a comment

rbolgaryn, 2020-11-06:
try setting the attribute "controlled_bus" of trafo_controller manually:

trafo_controller.controlled_bus = b3

Keep in mind, that the controller also enforces the limits of the tap position, and the resulting voltage will not match the set point that you specified. You can verify that it works by setting vm_set_pu to e.g. 1.03 or expanding the min/max tap limits of the transformer.

Roman

closed 2020-11-06 by rbolgaryn