e2nIEE/pandapower #977: State Estimation with disconnected Busses fails depending on the disconnecting switch

lennart-bader, 2020-11-06:
The state estimation fails as soon as the grid contains a disconnected bus. 

With the Cigre MV network as an example, when all line and trafo switches are closes, the estimation succeeds.
However, as soon as we decide to open switch 43 (at bus 8, disconnecting bus 19), the estimation fails to update the power grid.
Closing switch 43 and opening switch 42, which is on the same line, but at the side of bus 19, the estimation succeeds again.

So for disconnected "terminal" busses (i.e., a bus with a single line), the state estimation requires the line switch at this particular bus to be opened. For busses with multiple lines, we are unable to get an estimation result even when opening all related switches in different combinations.

Is this intended behavior, a missing feature, a bug, or does this behavior indicate a wrong usage from our side?

```python
pp.estimation.util.add_virtual_meas_from_loadflow(pnet)        

for key in pnet.keys():
    keysplit = key.split("_")
    if isinstance(pnet[key], DataFrame) and key[:3] == "res" and pnet[key].shape[0] and keysplit[-1] != "est":
        pnet[key].drop(pnet[key].index, inplace=True)

self.drop_zero_injection_bus_measurements(pnet)
self.drop_nan_measurements(pnet)   # Drops NaN values fromt he measurement rows

# Estimate
success = False
try:
    success = pp.estimation.estimate(pnet, init="flat", zero_injection="auto", maximum_iterations=1000)
finally:
    return success
```

wangzhenassd, 2020-11-06:
Hi,
can you offer the complete code to recreate the problem including the imports?

lennart-bader, 2020-11-06:
Hi, sorry for the delay - as we use a slightly modified CigreMV, I had to create a short example first.
In this example, we modify the default CigreMV network to have line switches on every side of each line.

The example script now first estimates the state with all switches closed (succeeds), then disconnects bus 6 by opening both line switches at bus 6 (succeeds again) and then disconnecting bus 6 by only opening the line switches at the other ends of the lines at bus 6 (fails).

After each change, we print the `switch` dataframe to show the changes.

This ist the test Script, the State Estimator Thread is below. I also attached the example module as ZIP.

```python
import pandapower as pp
import pandapower.networks as pn
from .state_estimator import StateEstimator
from threading import Event, Lock
import pandas

completed = Event()

def add_all_switches(net):
    net.switch.drop(net.switch.index, inplace=True)

    for line_id, row in net.line.iterrows():
        pp.create_switch(net, bus=row["from_bus"], element=line_id, et="l", closed=True)
        pp.create_switch(net, bus=row["to_bus"], element=line_id, et="l", closed=True)

    for trafo_id, row in net.trafo.iterrows():
        pp.create_switch(net, bus=row["hv_bus"], element=trafo_id, et="t", closed=True)
        pp.create_switch(net, bus=row["lv_bus"], element=trafo_id, et="t", closed=True)

    return net

def callback(success):
    global completed
    print(f"State Estimation {'succeeded' if success else 'failed'}")
    completed.set()

def main():
    cigremv = pn.create_cigre_network_mv()
    add_all_switches(cigremv)
    changed = Event()
    lock = Lock()

    pandas.set_option("display.max_rows", None)

    state_estimator = StateEstimator(cigremv, lock, changed, callback)
    state_estimator.start()

    with lock:
        cigremv.switch["closed"] = True # Close all switches
        pp.runpp(cigremv)

    print("Expected to succeed")
    changed.set()
    completed.wait()
    completed.clear()

    # Open Switches 24 and 9 (at bus 6)
    with lock:
        cigremv.switch.at[24, "closed"] = False
        cigremv.switch.at[9, "closed"] = False
        pp.runpp(cigremv)
        print(cigremv.switch)

    print("Usually succeeds")
    changed.set()
    completed.wait()
    completed.clear()

    # Open Switches at the other side of the lines
    with lock:
[... 178 lines cut ...]
            return False
        if switches is not None:
            for i, row in switches.iterrows():
                if not row["closed"]:
                    return False
        return True

    def drop_zero_injection_bus_measurements(self, net):
        injection_buses = set()
        for table in ["load", "shunt", "sgen", "gen", "ext_grid"]:
            injection_buses.update(net[table]["bus"])
        zero_injection_buses = list(set(net.bus.index) - injection_buses)
        df = net.measurement
        drop_ms = df[(df["measurement_type"].isin(["p", "q"])) & (df["element_type"] == "bus") & (
            df["element"].isin(zero_injection_buses))]

        print(f"will drop {len(drop_ms.index)} measurements of zero-injection buses")
        net.measurement.drop(drop_ms.index, inplace=True)
        pp.toolbox.close_switch_at_line_with_two_open_switches(net)

```
Here is also the full example module. Executable via `python3 -m se`.  
[se.zip](https://github.com/e2nIEE/pandapower/files/5501755/se.zip)

Thanks for your help!

wangzhenassd, 2020-11-06:
@lennart-bader I tried your "main" function locally, it failed in pandapower 2.2.2 but worked on pandapower >=2.3.0. What is your version of pandapower, can you try update it?

wangzhenassd, 2020-11-06:
I think it is fixed in #859 and released in 2.3.0