e2nIEE/pandapower #978: Can we set q_mvar for generators? It seems there are no q_mvar other than res_gen.

TravisCao, 2020-11-08:
I mean for the OPF, I want to compute the value of the Lagrangian function (generator dispatch cost plus the violation of constraints) of OPF given P and Q of the generators (we set it, not from the result of the runopp(net)). However, in the documentation, it seems that there is no way that we can set the Q for generators before the runopp() function is invoked. Can we set the Q (q_mvar) for generators?

rbolgaryn, 2020-11-09:
Hi @TravisCao ,

in general, the net.gen element is connected to a PV node, where the inputs are the active power P (p_mw) and set voltage V (vm_pu), and the outputs are the reactive power Q (q_mvar) and the voltage angle delta (va_degree). 

If you want to set a fix value for Q, you should use a PQ node where the inputs are P and Q, and the outputs are V and delta. You can use the element net.sgen for that.

If you want to have a PV node with a variable Q but have a constraint for the maximal or minimal Q, you can set the values of the constraints in net.gen.max_q_mvar, net.gen.min_q_mvar. In a load flow, pp.runpp, you can set the parameter "enforce_q_lims" to true. In an OPF, pp.runopf, the constraints should be considered by default.

Roman

friederikemeier, 2020-11-23:
I think, @rbolgaryn just explained it right 👍  If you want to set Q, use a static generator (sgen). I think this can be closed. Feel free to reopen if you have further questions.

closed 2020-11-23 by friederikemeier