e2nIEE/pandapower #990: Unknown mat file type, version 32, 99

15151874182, 2020-11-18:
Hi, when i try to use    pp_net=pandapower.converter.from_mpc('t_auction_case.m')

transfer a .mat net into a pandapower net, it gave me an error message:
Unknown mat file type, version 32, 99

do u have any ideas? thx

wangzhenassd, 2020-11-20:
Hi, can you share the complete error log?

15151874182, 2020-12-16:
> Hi, can you share the complete error log?

(grid2op) E:\cty\work_project\e-market-new>python
Python 3.6.10 |Anaconda, Inc.| (default, May  7 2020, 19:46:08) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pandapower
>>> pp_net=pandapower.converter.from_mpc('case24_ieee_rts.m')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\software\anaconda3\envs\grid2op\lib\site-packages\pandapower\converter\matpower\from_mpc.py", line 47, in from_mpc
    ppc = _mpc2ppc(mpc_file, casename_mpc_file)
  File "D:\software\anaconda3\envs\grid2op\lib\site-packages\pandapower\converter\matpower\from_mpc.py", line 55, in _mpc2ppc
    mpc = scipy.io.loadmat(mpc_file, squeeze_me=True, struct_as_record=False)
  File "D:\software\anaconda3\envs\grid2op\lib\site-packages\scipy\io\matlab\mio.py", line 223, in loadmat
    MR, _ = mat_reader_factory(f, **kwargs)
  File "D:\software\anaconda3\envs\grid2op\lib\site-packages\scipy\io\matlab\mio.py", line 72, in mat_reader_factory
    mjv, mnv = get_matfile_version(byte_stream)
  File "D:\software\anaconda3\envs\grid2op\lib\site-packages\scipy\io\matlab\miobase.py", line 231, in get_matfile_version
    raise ValueError('Unknown mat file type, version %s, %s' % ret)
ValueError: Unknown mat file type, version 82, 77

15151874182, 2020-12-16:
[case24_ieee_rts.m.zip](https://github.com/e2nIEE/pandapower/files/5699803/case24_ieee_rts.m.zip)
Here is my .m file, wish ur help, many thx

SteffenMeinecke, 2021-05-03:
The converter uses `.mat` files as source not `.m` files. Please run the `.m` script to get a variable named `mpc` into your matlab variable manager. Then save the workspace. Finally use the pandapower.converter.from_mpc(path_to_mat_file) functionality and link to the save `.mat` file.

SteffenMeinecke, 2021-07-21:
Close Issue due to no further problems stated

closed 2021-07-21 by SteffenMeinecke