e2nIEE/pandapower #992: Jacobian matrix after OPF run

piotre13, 2020-11-18:
dear all,
I'm quite new to this world and so please forgive me for the potential silly question:
I wanted to know if it is possible to extract the Jacobian matrix after the OPF run, I've seen the tutorial on the internal datastructure (based on Matpower), but actually neither in the _ppc['internal'] nor in the _ppc_opf['internal'] existe the key 'J'.
Thanks for the attention

lthurner, 2020-11-30:
I am not an OPF expert, but as far as I know the Jacobian Matrix is specific to the Newton-Raphson solver for the power flow. I don't think the OPF generates a Jacobian? Can you answer @friederikemeier ?

friederikemeier, 2020-12-03:
That's right.
But what you could do instead to "kind of" get the Jacobian for the current state after the opf is, that you write the results of the controllable elements into the input data tables and then run a normal runpp on that net, like this:

    net.sgen.p_mw = net.res_sgen.p_mw
    net.sgen.q_mvar = net.res_sgen.q_mvar
    # repeat this for all controllable elements, e.g. loads, gens
    runpp(net)
    J = net["_ppc"]['internal']

That's how I would do it.

closed 2021-11-11 by SteffenMeinecke