e2nIEE/pandapower #993: tapchanger controlling the voltage on a far busbar

mohammadswaiti, 2020-11-18:
how to control the voltage on a  busbar using the trafo tapchanger  and if this node is not beside the trafo in the following net
busbar 380kv - trafo 380kv/110kv - busbar 110kv - 50 km overhead line - busbar 'b'-load '80MVA' how can i controll the voltage on busbar 'b' to hold it on 110kv when i run powerflow 

`def createnetwork(net = 6):
    
    #leere Netz 
    net = pp.create_empty_network() 
    
     # Sammelschiene 380kv 
    b1 = pp.create_bus(net,vn_kv=380,type='b',name='bus1',in_service = True)
    
    # Sammelschiene 110kv
    b2 = pp.create_bus(net,vn_kv=110,type='b',name='bus2',in_service = True)
    
    # Sammelschiene 110kv
    b3 = pp.create_bus(net,vn_kv=110,type='b',name='bus3',in_service = True) 
    
    #Extragrid wird an den Sammelschiene1 verbunden 
    ext = pp.create_ext_grid(net, b1,voltage = 1.03,name = 'extra_grid',in_service = True)
    
#  trafo zwichen Sammelschien 1 '380kv' und Sammelschiene 2 '110kv' Scheinleistung '300MVA', Stufenschalter zwichen -9 und 9
#  stufenschalter seite ist an die Nidriege Spannung seite 
    trafo =pp.create_transformer_from_parameters(net, hv_bus=b1, lv_bus=b2, sn_mva=300, vn_hv_kv=380
                                                 , vn_lv_kv=110,name = 'Trafo1', vkr_percent=0.06, vk_percent=8,
                                                 shift_degree=1.5,
                                                 i0_percent=0.1,tap_side='lv',tap_phase_shifter= False, tap_neutral=0, 
                                                 tap_max=9, tap_min=-9,tap_step_percent= 2, tap_pos=x.value,pfe_kw=0,
                                                 in_service=True)

    #freileitung 50 km zwichen sammelschiene 2 und sammelschien 3  
    l1 = pp.create_line(net,b2,b3,name = '110kv line',length_km = 50.0,std_type = '184-AL1/30-ST1A 110.0',in_service = True)
    
    #Verbracher 
    load = pp.create_load(net,b3,p_mw=72,q_mvar=35, in_service=True)

    return net`

rbolgaryn, 2020-11-25:
Hi @mohammadswaiti ,

does this work for you?

https://github.com/e2nIEE/pandapower/issues/973#issuecomment-723074655

mohammadswaiti, 2020-12-02:
hi @rbolgaryn 

yes i have used the DiscreteTapControl and it worked 
thank you

closed 2020-12-02 by rbolgaryn