e2nIEE/pandapower #994: Drop support for Python 3.5?

rbolgaryn, 2020-11-18:
Python 3.5 is 5 years old now, and we can assume that most people already had the opportunity to upgrade to a more recent version. 

Some packages are already dropping support for python 3.5, e.g pandas >1 does not support python 3.5 anymore. Maintaining python 3.5 would mean an increased effort to include if-blocks for checking which pandas versions are installed, and dropping python 3.5 would reduce that sort of coding requirement somewhat.

We need a rule or best practice, which python versions to support and for how long.

Suggestion: drop python 3.5 support with the next release of pandapower.

rbolgaryn, 2020-11-30:
https://pandas.pydata.org/pandas-docs/stable/getting_started/install.html

Officially, pandas does not support 3.5. Pandas is essential for pandapower, so we need to drop support to 3.5 immediately.

jkupka, 2021-01-27:
With #1120 pandapower won't be tested anymore with Python 3.5 so official support for 3.5 is dropped.

closed 2021-02-03 by rbolgaryn