e2nIEE/pandapower #1004: calc_sc branch calculation error

apetros, 2020-11-20:
I have been investigating the calc_sc function with and without a fault impedance and met with a behavior I cannot easily explain. I know this is still in beta mode.

I used the tutorial on SC analysis from [here](https://github.com/e2nIEE/pandapower/blob/master/tutorials/shortcircuit.ipynb).

I compute the 3ph short circuit with fault impedance Z_F=0 and then again with Z_F=5+j10 Ohm. While the max fault currents at the nodes are decreasing as expected, the max fault currents on the branches are many times higher. 

As I understand, the currents are computed based on the post-fault voltages and the Ybus matrix. However, I expected when the short-circuit currents on each node are decreasing so would the max short-circuit current on the branches (roughly based on Kirchoff's law). Is this a bug or something I don't understand?

[image]

lthurner, 2020-11-26:
Thanks for reporting, this definitly looks like a bug. There is an open PR about SC branch currents #914. Maybe it already fixes the issue? I haven't gotten around to reviewing yet...

wangzhenassd, 2020-12-15:
I tried the example with the implementation on branch single_sc, it seems to work correctly. On develop, the bug can still be recreated.
[image]

lthurner, 2020-12-15:
Great, then I suggest integrating this as a test?

wangzhenassd, 2020-12-15:
> Great, then I suggest integrating this as a test?

Yes, i can do it

PR #1082 (2021-01-28): Update short circuit calculation 
According to #1037, the performance of short circuit for large networks are optimized:
1. added LU factorization mode instead of complete inversion of Ybus 
2. added bus selection and updated the result checkout
3. updated some tests
4. fixed #1004 
5. fixed #1064
6. Accroding to #78, a part of the example in iec60909 is now added
closed 2021-01-28 by lthurner