e2nIEE/pandapower #1008: const_control Definition matching_params

Pantherkralle, 2020-11-24:
Im const_controller wird in Zeile 56 die Variable matching params erstellt und anschließend ans parent übergeben. In auxiliary.py, Z. 171 wird dieses Argumement verarbeitet.

In const_control.py, Zeile 63 wird self.matching_params definiert. Hierbei wird allerdings nicht die Variable matching_params, die darüber definiert wurde, genutzt. Ist das bewusst so?

@friederikemeier

SteffenMeinecke, 2020-11-26:
Hey,
thank you very much for raising this issue. You are right, storing the parameter 'matching_params' to the controller in line 63 of const_control.py is unused. Therefore, we should either save the matching_params as it was used when creating the controller, or save no parameter self.matching_params at all.
For simplicity, I suggest not to store self.matching_params to controllers at all. 

Additionally, I have attached a sample script to show the expected function of 'matching_params':

```
import pandas as pd
import pandapower as pp
import pandapower.control as ct
import pandapower.networks as nw
import pandapower.timeseries as ts

net = nw.simple_mv_open_ring_net()
net.bus["zone"] = ["station"]*2 + ["feeder1"]*3 + ["Feeder2"]*2

data = pd.DataFrame({"p": [0.9, 1.1, 1.2],
                     "q": [0.15, 0.14, 0.18]})
ds = ts.DFData(data)

# add p-controller to buses [2, 3]
ctrl0 = ct.ConstControl(net, element="load", variable="p_mw", element_index=[2, 3],
                        profile_name="p", data_source=ds)

# the intersection of this p controller at bus 2 with buses [2, 3] of ctrl0 should be logged
ctrl1 = ct.ConstControl(net, element="load", variable="p_mw", element_index=2,
                        profile_name="p", data_source=ds)

# no log is needed with this q-controller at bus 2
ctrl2 = ct.ConstControl(net, element="load", variable="q_mvar", element_index=2,
                        profile_name="q", data_source=ds)

# a logging of q-controller at bus 3 gets necessary by giving shortend matching_params
ctrl3 = ct.ConstControl(
    net, element="load", variable="q_mvar", element_index=3, profile_name="q", data_source=ds,
    matching_params={"element": "load", "element_index": 3})

# replacing controller 'ctrl2' should be possible via 'drop_same_existing_ctrl'
ctrl4 = ct.ConstControl(net, element="load", variable="q_mvar", element_index=2,
                        profile_name="q", data_source=ds, drop_same_existing_ctrl=True)
print(net.controller)
```

rbolgaryn, 2020-12-02:
Hi @SteffenMeinecke ,

I have noticed, that the controller diagnostic function is not working as intended right now, and it seems that it is connected to the fact that ConstControl is not storing the attribute mathing_params.

The diagnostic shows warnings for ConstControl that are not warranted, and the cause is that the diagnostic function expexts the matching_params:

```python
    for idx in indices:
        current_controller = net.controller.object.loc[idx]
        parameters = deepcopy(current_controller.matching_params) if "matching_params" in \
                                                                     current_controller.__dict__ else {}
```

As you can see, if matching_params are not stored, the function falls back to an empty dictionary.

Roman

rbolgaryn, 2020-12-02:
Is there a way to add a test that checks if warnings are displayed?

Pantherkralle, 2020-12-02:
Hi,
just to be clear what I was thinking opening this issue:
The _matching_params_ given to the init function is processed in auxiliary.py, line 171 (I didn't find that, though, before Simon told me :).
However, the attribute _self.matching_params_ defined in line 63 isn't the same as defined in line 56 and given to the init function. Therefore, I was just wondering if it would cause problems when _matching_params_ is not _None_, because then, _self.matching_params_ would differ from _matching_params_ processed in auxiliary.py. Is that pretended to be like this?

SteffenMeinecke, 2020-12-08:
Sometimes one need to drop a line of code to find out what for it is needed - the control_diagnostic function is using `self.matching_params` - thank you @rbolgaryn.
With that knowledge, we can discuss the point of @Pantherkralle: Should `self.matching_params` be the same as given by the user or what is the default of each controller type?
Well, usually that is the same and if not I think we can assume that the user has a reason for giving special matching_params.
I'll fix it for the moment and we should keep the functionality of matching_params in mind when we discuss on moving controller information from the controllers variable space into `net.controller` (@SimonRubenDrauz).

closed 2021-07-07 by rbolgaryn