e2nIEE/pandapower #1009: short circuit, transformer impedance and tap changer "main position"

MacsBO, 2020-11-25:
Hi all, I think there is a problem with the calculation of parameters r and x of the transformers in case of short circuit.
I have a a 150 / 16.25 kV transformer that feed a 15 kV network and the short circuit value is higher than what I  expect

Looking at the code it seems to me that in the _calc_branch_values_from_trafo_df function of build_branch.py, while in case of power flow the parameters r and x of the transformers are calculated with reference to the effective transformation ratio, in case of short circuit they are calculated with reference to the rated transformation ratio.
I think it would be appropriate also for the short circuit to refer them to the transformation ratio in the "main position" (as IEC 60909 said) which could be significantly different from the nominal one.

what do you think?

lthurner, 2020-11-26:
Thanks for raising the issue. I remember that this was intentionally implemented this way since this was our interpretation of the IEC standard, and the results where also validated with commercial tools. However, I am a little fuzzy about the details.

Can you maybe put together a minimal example that showcases the issue with results and expected results?

rbolgaryn, 2020-11-27:
Hi @MacsBO ,

What exactly do you mean by the "main position"? What would be the main position for the transformer you are using in your example?

The transformer ratio is supposed to be accounted for using an adjustment factor c, which is defined in the function calc_sc depending on the case ("max" or "min").

The values of  c are set in _add_c_to_ppc(net, ppc) in build_bus.py. They are then used in the SC calculation to adjust for the possible difference of the tap position (e.g. see line 20 in currents.py), according to the standard IEC 60909. 

Does this behavior account for the effect you are describing?

MacsBO, 2020-11-27:
Hi all and thanks for the replies.
@rbolgaryn ,
in my example a 150 / 16.25 kV transformer that feed a 15 kV network have to work (without load) at +5 tap position to have 15 kV on secondary winding. so for me the "main position" is +5.
If the transformer had been a 150/15 kV, the "main position" would have been 0.

@lthurner
I will prepare a simple example as soon as possible and share it with you

lthurner, 2020-12-08:
That is not my understanding of the main position. For me, the position 0 would be the main position per definition.

If we would always calculate the main position from the nominal transformer ratio, that would render the rated transformer voltages meaningless, because a 150/16.25kV transformer would behave exactly the same as the 150/15kV transformer.

Can you point to the section in IEC 60909 that you believe supports your definition of the main position?

rbolgaryn, 2021-06-01:
Hi @MacsBO ,

do you have any updates with this issue?

Roman

MacsBO, 2021-06-03:
Hi @rbolgaryn , sorry but I've been very busy and haven't written the sample network yet. I hope to finish it quickly. By checking the results of the short-circuit calculation, however, it seems to me that there is a correction factor related to the relationship between the nominal mains and winding voltages, even if the results are not what I expected.