e2nIEE/pandapower #1013: Computation speed

hitchenzhe, 2020-11-25:
Dear all,

I used power flow computation for my model, it is a simple 110 buses network from simbench. And I input time series data from my optimization part. The 1 year simulation with hourly data cost 20 min. Any idea to speed up the load flow? 

How about Pypower? Do they use the same core?

Best,

wangzhenassd, 2020-11-26:
pandapower has actually performance optimized against the default pypower PF solver based on numba. Can you check if numba is installed?

hitchenzhe, 2020-11-26:
> pandapower has actually performance optimized against the default pypower PF solver based on numba. Can you check if numba is installed?

Yes, I installed. I am thinking how to increase the computation speed.
`
(base) C:\Users\zhech>pip install numba
Requirement already satisfied: numba in c:\users\zhech\anaconda3\lib\site-packages (0.50.1)
Requirement already satisfied: llvmlite<0.34,>=0.33.0.dev0 in c:\users\zhech\anaconda3\lib\site-packages (from numba) (0.33.0+1.g022ab0f)
Requirement already satisfied: numpy>=1.15 in c:\users\zhech\anaconda3\lib\site-packages (from numba) (1.18.5)
Requirement already satisfied: setuptools in c:\users\zhech\anaconda3\lib\site-packages (from numba) (49.2.0.post20200714)
`

hitchenzhe, 2020-11-30:
> pandapower has actually performance optimized against the default pypower PF solver based on numba. Can you check if numba is installed?

One question, how to see whether I enable numba in pandapower?

rbolgaryn, 2020-11-30:
Hi @hitchenzhe ,

it should be enabled by default, if you installed it.
Numba saves compiled functions to pandapower\pandapower\pf\__pycache__ and you can check if that's the case for you.

Roman

hitchenzhe, 2020-11-30:
> Hi @hitchenzhe ,
> 
> it should be enabled by default, if you installed it.
> Numba saves compiled functions to pandapower\pandapower\pf__pycache__ and you can check if that's the case for you.
> 
> Roman

Thank you , do you know how to set linearized option in time series load flow?

rbolgaryn, 2020-11-30:
What do you mean by linearized option?

rbolgaryn, 2020-11-30:
If you can pass an option to pp.runpp, you can also set it in net.user_pf_options. You can use the function pp.set_user_pf_options(net, option='option'). In that case, it will be considered every time you run load flow simulation, even if you don't explicitly pass it.

hitchenzhe, 2020-11-30:
> What do you mean by linearized option?

Thank you, I mean the linearized load flow or we can say DC load flow. So in the time series function, it is default to use runpp() with **kwargs. I don know how to change to dccpp when we run time series.

run_timeseries(net, time_steps=None, continue_on_divergence=False, verbose=True, **kwargs)

rbolgaryn, 2020-12-01:
you can see in the docstring for the function pandapower.timeseries.run_timeseries:

Keyword arguments for run_control and runpp. If "run" is in kwargs the default call to runpp() is replaced by the function kwargs["run"]

closed 2020-12-02 by rbolgaryn