e2nIEE/pandapower #1016: OPF fails in simbench nets

thomaswolgast, 2020-11-29:
The pandapower OPF fails to converge in in simbench nets in pandapower version 2.3 and 2.4 (2.2 is working). Since I only changed version number of pandapower, I am quite sure this is a bug in the OPF.

Here is a code example to reproduce the error: (test with pandapower==2.3 and pandapower==2.2)
```
import simbench as sb
import pandapower as pp

# net = sb.get_simbench_net('1-MV-rural--0-sw')
# net = sb.get_simbench_net('1-MV-semiurb--0-sw')
net = sb.get_simbench_net('1-MV-urban--0-sw')

net.bus['min_vm_pu'] = 0.95
net.bus['max_vm_pu'] = 1.05
net.line['max_loading_percent'] = 100
net.trafo['max_loading_percent'] = 100
net.trafo['tap_pos'] = 1

profiles = sb.get_absolute_values(net, profiles_instead_of_study_cases=True)

for elm_para in profiles.keys():
    if profiles[elm_para].shape[1]:
        net[elm_para[0]].loc[:, elm_para[1]] = profiles[elm_para].loc[1]

net['sgen']['controllable'] = True
net['sgen']['min_p_mw'] = 0
net['sgen']['max_p_mw'] = net['sgen']['p_mw']
net['sgen']['min_q_mvar'] = -net['sgen']['max_p_mw'] * 0.3
net['sgen']['max_q_mvar'] = -net['sgen']['min_q_mvar']

pp.runopp(net)
```

=> `pandapower.optimal_powerflow.OPFNotConverged: Optimal Power Flow did not converge!`

I could reproduce the error for three different simbench nets (all I tested) and for python versions 3.7 and 3.8. Maybe in newer version different settings are required? But even then, I would expect a different error message instead of not converged.

thomaswolgast, 2020-12-02:
I had a similar problem (see issue #1016). If you try the same thing with pandapower 2.2, does it work then?

rbolgaryn, 2020-12-02:
possibly connected to #1020

friederikemeier, 2020-12-02:
That is indeed weird. What I would do is to check the difference between the net["_ppc"] tables between 2.2 and 2.4.

- Install pp 2.4

- run the OPF

- save net["_ppc"]

- Install pp 2.2

- run the OPF

- save net["_ppc"]

And compare. We may have changed default values somewhere causing this. But I don't remember at the moment. If you post the diff of the ppcs, please tag me and I'll have a look again.

joshuajakob, 2021-02-02:
@thomaswolgast Meanwhile, I have the same problem. Is there already a solution for it?

SteffenMeinecke, 2021-11-11:
I checked the OPF problem by solving it with another method than runopp, which is a usage of the pypower OPF implementation: The net configuration and the OPF problem are fine. As expected before, the issue is in the convergence behaviour of the pypower OPF implementation, which is poor. Thats the reason why we started working with other methods long ago, such as powermodel.
Sorry to say but I think here we can not do anything obvious -> close issue.

closed 2021-11-11 by SteffenMeinecke