e2nIEE/pandapower #1020: DC OPF does not converge

andrea-belle, 2020-11-30:
I want to perform a DC OPF in a situation of high disruption using the GB reduced network, but sometimes it does not converge. My cost function is linear, with cp1 = 1 for each generator and cp1 = -100 for each load. I report here an example with the code. If I assume to fail the buses 2, 4, 11, 12, 20 and 25, the OPF does not converge. From how I see it, it should always converge, since a trivial solution always exists (set the power of loads and generators to 0). I noticed this behaviour also with other networks. Any clue on what might be the problem?
`import pandapower as pp

net = pandapower.networks.GBreducednetwork()
     
net.line.loc[:,'max_loading_percent'] = 100
net.trafo.loc[:,'max_loading_percent'] = 100

net.load.loc[:,'min_p_mw'] = 0
net.load.loc[:,'max_p_mw'] = net.load.p_mw

net.load.loc[:,'controllable'] = True

net.poly_cost.loc[:,'cp0_eur'] = 0
net.poly_cost.loc[:,'cp1_eur_per_mw'] = 1
net.poly_cost.loc[:,'cp2_eur_per_mw2'] = 0

for load in list(net.load.index):
    globals()['costload%s' %load] = pp.create_poly_cost(net, load, 'load',cp0_eur=0,
                                                        cp1_eur_per_mw = -100,cp2_eur_per_mw2=0)

failed_bus = [2, 4, 11, 12, 20, 25]
pp.set_element_status(net, failed_bus, in_service=False)
pp.rundcopp(net)`

christoph-weyhing, 2020-12-01:
I am facing similar issues when I try to run a DCOPF with some Simbench networks. I set the controllable attribute to 'True' for all generators (gen), static generators (sgen) and loads and assigned linear cost functions, but in some timesteps the OPF does not converge, even if there is always a trivial solution. I am using pandapower 2.4.

thomaswolgast, 2020-12-02:
I had a similar problem (see issue #1016). If you try the same thing with pandapower 2.2, does it work then?

rbolgaryn, 2020-12-02:
I can reproduce it with DC OPF, and OPF converges with AC OPF.

andrea-belle, 2020-12-02:
I tried with versions 2.1.0, 2.2.0, 2.3.0 and 2.4.0, and they all give me the same error. I am using Python 3.8, maybe I could try with 3.7 and 3.6, but I guess the problem is not there. I did not try with an AC OPF, in my case I really need a DC OPF.

rbolgaryn, 2020-12-02:
Unfortunately, we will not have the time to look into this issue until January.

rbolgaryn, 2020-12-02:
You can try using the OPF via the interface to PowerModels. The interface is described in the tutorial in the folder path tutorials/opf_powermodels.ipynb

rbolgaryn, 2020-12-02:
Please let us know if it works

andrea-belle, 2020-12-02:
I will try probably in the next weeks and let you know if it works. Something that I forgot to mention: if I increase the linear cost cp1_eur_per_mw of the loads, it might converge. For example, in this specific case, it converges if I set it above -4.

andrea-belle, 2020-12-02:
I noticed the same behaviour with other networks and other combinations of failures. Sometimes, changing the cost of the loads is sufficient for having a convergence (not always). I do not really understand why, since there should always be a solution.

andrea-belle, 2020-12-02:
I tried with the PowerModel pp.runpm_dc_opf(net) and it worked.
Since I am not familiar with Julia, I take advantage of asking a trivial question. Every time I run pp.runpm_dc_opf(net), I have a sort of loop printing "WARNING: replacing module PP2PM." and it is quite annoying. Do you know how to get rid of it? 
I am using Python 3.8 installed with Anaconda.

rbolgaryn, 2020-12-03:
> 
> 
> I will try probably in the next weeks and let you know if it works. Something that I forgot to mention: if I increase the linear cost cp1_eur_per_mw of the loads, it might converge. For example, in this specific case, it converges if I set it above -4.

It is interesting, maybe it is caused by numerical issues. Sometimes changing net.sn_mva affects convergence, maybe there is a similar effect in this case.

rbolgaryn, 2020-12-03:
> 
> 
> I will try probably in the next weeks and let you know if it works. Something that I forgot to mention: if I increase the linear cost cp1_eur_per_mw of the loads, it might converge. For example, in this specific case, it converges if I set it above -4.

You can also try adjusting the tolerances of the OPF, you can look in the function definition which parameters refer to tolerances.

MMajidi137, 2020-12-03:
> I tried with the PowerModel pp.runpm_dc_opf(net) and it worked.
> Since I am not familiar with Julia, I take advantage of asking a trivial question. Every time I run pp.runpm_dc_opf(net), I have a [...]

Hi, thanks for informing us. This warning is not related to the PowerModel.jl package, and it does not affect the result of OPF. PP2Pm is our internal module in our interface from pandapower to Powermodels.  This warning is about how we call and use the module in our interface is needed to change in a different way.  we will work on it.

andrea-belle, 2020-12-03:
> Hi, thanks for informing us. This warning is not related to the PowerModel.jl package, and it does not affect the result of OPF. PP2Pm is our internal module in our interface from pandapower to [...]

Thanks for the answer. I am running the OPF in a loop. After a few cycles, it takes a lot of time just to print all the warnings. Do you have any suggestion on how to bypass this problem?

MMajidi137, 2020-12-03:
> Thanks for the answer. I am running the OPF in a loop. After a few cycles, it takes a lot of time just to print all the warnings. Do you have any suggestion on how to bypass this problem?

are all about replacing the module?

andrea-belle, 2020-12-03:
> are all about replacing the module?

Yes, at each new cycle an extra PP2PM warning is printed, except for the very first one. After 10 cycles, 9 warnings are printed and so on..

MMajidi137, 2020-12-03:
> Yes, at each new cycle an extra PP2PM warning is printed, except for the very first one. After 10 cycles, 9 warnings are printed and so on..

please follow this to suppress warnings from Julia:
``import pandapower  as pp
import pandapower.networks as nw
from julia.api import Julia

Julia(warn_overwrite=False) 
net = nw.example_simple()
pp.runpm_dc_opf(net)``

friederikemeier, 2020-12-03:
Hi!
You could also try to play with the tolerances:

         - delta (1e-10)
         - OPF_VIOLATION (5e-6) constraint violation tolerance
         - PDIPM_COSTTOL (1e-6) optimality tolerance
         - PDIPM_GRADTOL (1e-6) gradient tolerance
         - PDIPM_COMPTOL (1e-6) complementarity condition (inequality) tolerance
         - PDIPM_FEASTOL (set to OPF_VIOLATION if not specified) feasibiliy (equality) tolerance
         - PDIPM_MAX_IT  (150) maximum number of iterations
         - SCPDIPM_RED_IT(20) maximum number of step size reductions per iteration

I also suppose that it's a numerical issue.

andrea-belle, 2020-12-03:
> please follow this to suppress warnings from Julia:
> ``import pandapower as pp
> import pandapower.networks as nw
> from julia.api import Julia
>
> Julia(warn_overwrite=False)
> net = nw.example_simple() [...]

I tried but it does not work for me. If I run your code three times, these are the three outputs:

1st time:
```
runfile('C:/Users/Andrea/Desktop/power_rail/forum_problem.py', wdir='C:/Users/Andrea/Desktop/power_rail')
no costs are given - overall generated power is minimized
```
2nd time:
```
runfile('C:/Users/Andrea/Desktop/power_rail/forum_problem.py', wdir='C:/Users/Andrea/Desktop/power_rail')
no costs are given - overall generated power is minimized

WARNING: replacing module PP2PM.
```
3rd time
```
runfile('C:/Users/Andrea/Desktop/power_rail/forum_problem.py', wdir='C:/Users/Andrea/Desktop/power_rail')
no costs are given - overall generated power is minimized

WARNING: replacing module PP2PM.
WARNING: replacing module PP2PM.
```

The outputs above are from Spyder. If I run the same code in the Anaconda PowerShell prompt, I don't get the warnings but the execution of the script is very slow, like if it was opening Julia in background at every different execution of the script. With Spyder, this happens only the very first execution, like it is supposed to be I guess.

christoph-weyhing, 2020-12-08:
> please follow this to suppress warnings from Julia:
> ``import pandapower as pp
> import pandapower.networks as nw
> from julia.api import Julia
>
> Julia(warn_overwrite=False)
> net = nw.example_simple() [...]

I am facing the same issue too. If I run
`run_timeseries(net, time_steps, continue_on_divergence=True, verbose=True, recycle=False, run=pp.runpm_dc_opf, delta=1e-12)`
I get a list of warnings between each progress bar that says "WARNING: replacing module PP2PM". When I use
`from julia.api import Julia
 Julia(warn_overwrite=False)`
it doesn't change anything. The warnings still appear.

friederikemeier, 2020-12-11:
Hi! Just a quick update on this: @MMajidi137 has a solution for this and will push some updates to solve this in the beginning of 2021. The problem is, that we don't have our stuff defined as a "real" julia package. 
Thanks for all of your feedback and input :)

andrea-belle, 2021-01-15:
> Hi! Just a quick update on this: @MMajidi137 has a solution for this and will push some updates to solve this in the beginning of 2021. The problem is, that we don't have our stuff defined as a "real" julia package.
> Thanks for all of your feedback and input :)

Hello. I saw that version 2.5.0 has been released, but the Julia "warning" problem still persists. Do you plan to solve it in the next release? Thanks in advance.

SteffenMeinecke, 2021-11-11:
@MMajidi137 seems that the issues to the powermodels bridge are addressed, right?
A improvement of the pypower OPF convergence behaviour is not planned, compare to #1016.
Close this issue?

MMajidi137, 2021-11-11:
@SteffenMeinecke yes we can close this issue now

closed 2021-11-11 by SteffenMeinecke