e2nIEE/pandapower #1021: Inconsistent 3ph power flow results for transformers

meyer89, 2020-11-30:
I use pandapower 2.4 for 3ph asymmetric power flows and encountered some strange results for transformers.

Steps to reproduce:
```python
from pandapower.test.loadflow.test_runpp_3ph import make_nw, pp
net = pp.create_empty_network(sn_mva=100)
make_nw(net, 10, 0, "wye", "Dyn")
pp.runpp_3ph(net)
print(net.res_trafo_3ph[["i_%s_lv_ka" % phase for phase in "abc"]])
print(net.res_line_3ph[["i_%s_from_ka" % phase for phase in "abc"]])
```

On my computer, this generates the following results:

|    |   i_a_lv_ka |   i_b_lv_ka |   i_c_lv_ka |
|----|-------------|-------------|-------------|
|  1 |    0.136207  |  0.130348   | 0.129255 |

|    |   i_a_from_ka |   i_b_from_ka |   i_c_from_ka |
|----|---------------|---------------|---------------|
|  1 |      0.14161    |  0.129326      | 0.124782 |

The example network is simple with 1 transformer, 1 line, 1 asymmetric_load and 3 buses. The bus 1 has only the line and transformer connected.

The current values *should* match, but the difference is significant.

Only when the asymmetric loads are symmetric, the calculated current values of line and transformer are identical.
```python
net.asymmetric_load[["p_%s_mw" % phase for phase in "abc"]] = 0.03
net.asymmetric_load[["q_%s_mvar" % phase for phase in "abc"]] = 0.0048
pp.runpp_3ph(net)
print(net.res_trafo_3ph[["i_%s_lv_ka" % phase for phase in "abc"]])
print(net.res_line_3ph[["i_%s_from_ka" % phase for phase in "abc"]])
```

|    |   i_a_lv_ka |   i_b_lv_ka |   i_c_lv_ka |
|----|-------------|-------------|-------------|
|  1 |    0.140402 |    0.140402 |    0.140402 |

|    |   i_a_from_ka |   i_b_from_ka |   i_c_from_ka |
|----|---------------|---------------|---------------|
|  1 |      0.140402 |      0.140402 |      0.140402 |

Can you reproduce the problem? Right now there seems to be no check of the transformer results within the test suite.

lthurner, 2020-12-08:
Thanks for reporting, that is indeed strange. The voltages seem to be correct, but the currents are incosistent. I hope that its just a lookup or transformation issue in either the line or transformer element. If you could look into it yourself, that would be much appreciated.

meyer89, 2020-12-08:
I agree that the voltages are correct. Also line currents should be correct, as they match the results from tools like PowerFactory. 

The complex currents for the transformer based on power flow and complex voltage for each phase have the same magnitude as the results for i_[abc]_lv_ka in res_trafo_3ph. Therefore, the power values are most likely also inconsistent.

The sum of the three complex currents i_[abc]_lv_ka is always zero, even when loads are not symmetric and there is a current in the neutral conductor of the line. My calculation is validated on the line and it matches the result of i_n_ka in res_line_3ph.

Maybe I have the time to look into this issue a little bit deeper.

rbolgaryn, 2021-02-24:
I think this could be related to #1073

lthurner, 2021-04-06:
Thanks for suggesting a solution for this problem @elodin. As I have explained in more detail in #1180, I don't believe the proposed solution is correct, but it did give me an idea what the problem might be.

The transformer impedances in the zero sequence are added to the bus matrix rather then the branch matrix, because the impedances are between HV/LV and ground rather than between HV and LV (for reference, see the [AESO Tranformer Modeling Guide](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwiE1quW8-nvAhVq_rsIHUk1AzcQFjAAegQIBRAD&url=https%3A%2F%2Fwww.aeso.ca%2Fassets%2Flinkfiles%2F4040.002-Rev02-Transformer-Modelling-Guide.pdf&usg=AOvVaw16YyJb_H4h5ZQV0NNWUOzR), page 51). The pandapower result extraction assumes all transformer impedances are in the ppci branch table, and calculates the currents from there. However, the zero sequence transformer impedances are in the bus table, and therefore not considered in the transformer current calculation. That is probably why the voltages are fine (because the impedances in the matrix are correct), but the currents are wrong (because the result calculation doesn't recognize the transformer impedances in the bus table as part of the transformer).

lthurner, 2021-07-09:
Any progress on this issue @elodin @gourab2009 ?

elodin, 2021-07-12:
I got a bit enganged to shape my master thesis and meeting a few deadlines. I am again trying to bring my focus back to this particular topic in parallel to my thesis. 
I am looking into this matter and trying to resolve this issue in the coming weeks. Gourab @gourab2009 is also giving his support again to me to go further with this problem.
So far it seems to me, that the currents are calculated using voltages and powers. So I will look into the power calculation to see where the impedances of the bus table can be added in the right way.

elodin, 2021-07-24:
I am a bit confused about the units of shunt conductance (Gs) and shunt susceptance (Bs) that are saved in the bus table and can be used to calculate an admittance/impedance. I would expect them to be measured in Siemens, however the idx_bus.py says the following:
GS = 4    # Gs, shunt conductance (MW at V = 1.0 p.u.)
BS = 5    # Bs, shunt susceptance (MVAr at V = 1.0 p.u.)
Do i misunderstand something or might the comment be wrong here?

elodin, 2021-11-19:
@SteffenMeinecke @rbolgaryn @SimonRubenDrauz
This issue has been addressed together with Gourab. The solution is giving correct results and needs to be checked for merging. Please tell me if it should be revised in any way or else it can be merged.

SteffenMeinecke, 2022-01-10:
Thanks @elodin, @gourab2009 still needs to check the changes. Sorry for the time delay.

closed 2022-01-10 by SteffenMeinecke