e2nIEE/pandapower #1026: Options in time series load flow

hitchenzhe, 2020-12-01:
Hi,

Anyone tell me how to set different options in load flow series simulation:
pandapower.timeseries.run_time_series.run_timeseries(net, time_steps=None, continue_on_divergence=False, verbose=True, **kwargs)

for example, if I want to use runcpp with different tolerances, set angle or not, enable numba or not, and use rundcpp or not.

Thanks in advances

rbolgaryn, 2020-12-01:
Hi @hitchenzhe ,

you can use pp.set_user_pf_options(net, key=value)

hitchenzhe, 2020-12-01:
> Hi @hitchenzhe ,
> 
> you can use pp.set_user_pf_options(net, key=value)

Thanks, another question is do we have an option for DistFlow or LinDistFlow model in load flow calculation. As far as I know, pandapower uses full non-linear equations.

Best,

rbolgaryn, 2020-12-02:
Hi @hitchenzhe ,

I don't think so.
Would you be interested in imoplementing it?

Roman

hitchenzhe, 2020-12-02:
> Hi @hitchenzhe ,
> 
> I don't think so.
> Would you be interested in imoplementing it?
> 
> Roman

Yes, I am thinking how to implement it. May I ask that what load flow model used in pandapower? is it DistFlow model?

rbolgaryn, 2020-12-02:
Hi @hitchenzhe ,

the power flow equations are the same as in Matpower, you can read the formulations of the power flow equations and the  in their documentation. It is a standard formulation that can be found in literature.

The pandapower power flow solver is based on the Newton-Raphson method. The implementation was originally based on PYPOWER, but has been improved with respect to robustness, runtime and usability. You can also refer to the pandapower documentation:

https://www.pandapower.org/about/#pf
https://www.pandapower.org/about/#modeling
https://pandapower.readthedocs.io/en/stable/powerflow/ac.html

hitchenzhe, 2020-12-02:
> Hi @hitchenzhe ,
>
> the power flow equations are the same as in Matpower, you can read the formulations of the power flow equations and the in their documentation. It is a standard formulation that [...]

Thank you for the replying. Helpful !
Then I go to OPF part. Some questions I didn find in doc.
1. What solver you used to solved optimization? (any option to change solver?)

2. When we have load flow constraints, how pandapower formulates it? Since the standard form of pf is non convex. How can we guarantee global optimum.

3. Any option to remove some constraints? For example, if I don need line loading constraint, how can I remove it ?

rbolgaryn, 2020-12-02:
1) The implementation is pypower OPF. In our work, we also use other solvers, e.g. scipy linsolve to solve LP-problems. The benefit would be better convergence if pypower OPF does not converge. You can also use the PowerModels interface (some examples available in tutorials). 

2) I would refer to the documentation of matpower OPF for this point. 

3) For your case, you could set the line loading limit to 1000 %. A more flexible definition of constraints is possible with PowerModels, with the pandapower-PowerModels interface.

hitchenzhe, 2020-12-03:
> 1. The implementation is pypower OPF. In our work, we also use other solvers, e.g. scipy linsolve to solve LP-problems. The benefit would be better convergence if pypower OPF does not converge. You [...]

Thank you !

How about PTDF/LODF calculation? Do we have any command for this?

rbolgaryn, 2020-12-03:
Take a look in the directora pandapower.pypower, there is an impolementation from pypower

closed 2020-12-04 by rbolgaryn