e2nIEE/pandapower #1033: Scalability of pandapower.shortcircuit.calc_sc

bergkvist, 2020-12-03:
## Memory consumption
Right now, `pandapower.shortcircuit.calc_sc` seems to require O(n^2) memory for storing the impedance matrix when you have n buses in your pandapower network. This is because it stores the impedance between every pair of buses.

The values in the impedance matrix are of type `complex128` - which means every value takes up 128 bits (16 bytes).

| # of buses | memory needed to store impedance matrix |
|-|-|
| 1 | ~16 B |
| 10 | ~1.56 KiB |
| 100 | ~156 KiB |
| 1000 | ~15.3 MiB |
| 10,000 | ~1.49 GiB |
| 100,000 | ~149 GiB |
| 1,000,000 | ~14.6 TiB |

Basically, the current implementation of calc_sc stops being usable after 10,000 buses.

## Unnecessary calculation and memory usage
So are all of these values actually used? My understanding is that only the impedance between ext_grid/gens and the rest of the buses are relevant.

Assuming you have a single ext_grid, and no generators - the part of the impedance matrix that is actually used for calculation would only be O(n); A single row/column of the impedance matrix.

In my head, the common case is that you have very few generators compared to the number of buses in your network. Which means avoiding calculation/storage of bus-bus impedance that are not used could have a huge impact on performance/scalability.

rbolgaryn, 2020-12-03:
Hi @bergkvist ,

can it also be that a dense matrix is used instead of a sparse matrix at some point?

wangzhenassd, 2020-12-03:
Yes, the inverse of the Ybus would be a dense matrix

lthurner, 2020-12-03:
The short-cicuit currents are calculated for every bus in the network in one calculation. This requires the invertion of the Ybus matrix, which is dense as @wangzhenassd mentioned. A reduction of the branches would be possible, but then the short-circuit currents would not be available at the buses that are reduced (which might be acceptable?). I don't know of a different way to calculate this problem more efficiently.

If a short-circuit should only be calculated at one bus, it would be possible to do this without inverting the Ybus matrix. There already is a function that calculates a single short-circiuit current: sc.calc_single_sc. But it also relies on the full invertion of the matrix. Refactoring this to solving an equation system instead of inverting the matrix is possible and on our to do list.

bergkvist, 2020-12-03:
Illustration of what I talked about above:
[image]

lthurner, 2020-12-03:
Actually we only ever need the diagonal of the ZBus matrix. But to get the diagonal, we need calculate the Zbus matrix (invert the Ybus matrix).

What you are suggesting only works for a very limited use case. You have a non-meshed grid and only one generator. You could write a function that is more efficient and only works with this limited way. I don't see a way to improve the generalized function that calculates SC currents for all kind of network topologies at all buses though.

bergkvist, 2020-12-03:
The diagonal? In that case I think I've misunderstood the layout of the ZBus matrix.

My view was that Zbus[x,y] represented the impedance between bus x and y. In which case Zbus[x,x] would just be 0? (No impedance between a bus and itself)

lthurner, 2020-12-03:
No I don't think thats the case. I have detailed the SC calculation equations here: https://pandapower.readthedocs.io/en/v2.4.0/shortcircuit/ikss.html#voltage-source-contribution. That should make clear why its the diagonal

lthurner, 2020-12-15:
See #1037 and development on the single_sc branch

closed 2020-12-15 by lthurner