e2nIEE/pandapower #1037: Add a faster version for shortcircuit calculation for single bus

wangzhenassd, 2020-12-04:
Please checkout branch single_sc

wangzhenassd, 2020-12-04:
A simple example
```python
from copy import deepcopy
import pandapower as pp
from pandapower.shortcircuit import calc_sc, calc_single_sc
import pandapower.networks as nw

net = nw.create_cigre_network_mv()
net.ext_grid['s_sc_max_mva'] = 1000
net.ext_grid['rx_max'] = 0.1
net.sgen = net.sgen.iloc[0:0, :]

calc_single_sc(net,5)        
calc_single_sc(net_no_inv, 5, with_y_inv=False)
    
print(net.res_bus)
print(net.res_bus_sc)
```

wangzhenassd, 2020-12-04:
It also seems to work with larger system:
```
    net = nw.case1354pegase()    
    net.ext_grid['s_sc_max_mva'] = 1000
    net.ext_grid['rx_max'] = 0.1
    net.gen["vn_kv"] = net.bus.loc[net.gen.bus.values, "vn_kv"].values
    net.gen["rdss_pu"] = 0.01
    net.gen["xdss_pu"] = 0.1
    net.gen["cos_phi"] = 0.8
    net.gen["sn_mva"] = net.gen.p_mw.values
    
    net.sgen['k'] = 1.
    net.sgen = net.sgen.iloc[0:0, :]

    net_no_inv = deepcopy(net)

    calc_single_sc(net,254)        
    calc_single_sc(net_no_inv, 254, with_y_inv=False)

    print((net.res_bus_sc - net_no_inv.res_bus_sc).max())
```

bergkvist, 2020-12-10:
It seems like the `calc_single_sc` is not yet implemented for 1ph-short circuits. This is also something that would be nice to have.

#### `calc_sc`:
https://github.com/e2nIEE/pandapower/blob/c9aaecba98c6a0c812452a82fb7064b82bde1c70/pandapower/shortcircuit/calc_sc.py#L128-L133

#### `calc_single_sc`:
https://github.com/e2nIEE/pandapower/blob/c9aaecba98c6a0c812452a82fb7064b82bde1c70/pandapower/shortcircuit/calc_sc.py#L198-L201

wangzhenassd, 2021-01-04:
#1082 should fulfil the first phase of the improved short circuit calculation functionality

TODO:

- [ ] Discuss and maybe remove original single_sc calculation
- [ ] Add ib calculation on bus for generator-close short calculation
- [ ] Further updates in test

wangzhenassd, 2021-01-04:
> It seems like the `calc_single_sc` is not yet implemented for 1ph short-circuits. This is also something that would be nice to have.
>
> #### `calc_sc`: [...]

Hi,
I think after #1082 merged, the functionality is also fulfilled.

wangzhenassd, 2021-01-04:
@lthurner @rbolgaryn Do we want to maybe remove the original calc_single_sc function? since in PR the calc_sc can already define the required bus and also use the fast version.

lthurner, 2021-02-18:
Yes calc_single_sc can be removed @wangzhenassd

PR #1137 (2021-05-26): Multiple functional improvements in shortcircuit calculation
1. Added support generator-close sc calculation as well as on generation buses
2. Added support of power station calculation for OLTC and non-OLTC transformer (needs to be checked)
3. Removed single_sc calculation, since it is now integrated in calc_sc
4. Updated the internal program flow of calc_sc
2. Initialized the calculation of 1ph shortcircuit calculation with trafo3w in network
3. Added test case according to IEC 60909-4, which should close #78
4. Changed the convention of rdss_pu to rdss_ohm to generator, since in IEC 60909 r [... cut]
closed 2021-05-26 by lthurner