e2nIEE/pandapower #1039: Load Flow configuration questions

hitchenzhe, 2020-12-04:
Hello,
This is my code and it return an error:(I have installed pypower)

pp.set_user_pf_options(net, algorithm = 'fdbx', max_iteration=10, calculate_voltage_angles=0, tolerance_mva=1e-04)
run_timeseries(net, time_steps)
`
pp.set_user_pf_options(net, algorithm = 'fdbx',max_iteration=10, calculate_voltage_angles=0, tolerance_mva=1e-04)
run_timeseries(net, time_steps)
Progress: |--------------------------------------------------| 0.0% CompleteTraceback (most recent call last):

  File "<ipython-input-5-00e001fa95c8>", line 2, in <module>
    run_timeseries(net, time_steps)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\timeseries\run_time_series.py", line 338, in run_timeseries
    run_loop(net, ts_variables, **kwargs)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\timeseries\run_time_series.py", line 310, in run_loop
    run_time_step(net, time_step, ts_variables, **kwargs)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\timeseries\run_time_series.py", line 114, in run_time_step
    run_control(net, run_control=False, ctrl_variables=ts_variables, **kwargs)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\control\run_control.py", line 234, in run_control
    control_implementation(controller_order, net, run_funct, errors, max_iter, continue_on_lf_divergence, **kwargs)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\control\run_control.py", line 152, in control_implementation
    _control_repair(net, run_funct, continue_on_lf_divergence, levelorder, errors, **kwargs)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\control\run_control.py", line 172, in _control_repair
    run_funct(net, **kwargs)  # run can be runpp, runopf or whatever

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\run.py", line 225, in runpp
    _powerflow(net, **kwargs)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\powerflow.py", line 74, in _powerflow
    result = _run_pf_algorithm(ppci, net["_options"], **kwargs)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\powerflow.py", line 153, in _run_pf_algorithm
    result = _runpf_pypower(ppci, options, **kwargs)[0]

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\pf\runpf_pypower.py", line 58, in _runpf_pypower
    ppci, success, bus, gen, branch, it = _ac_runpf(ppci, ppopt, numba, recycle)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\pf\runpf_pypower.py", line 95, in _ac_runpf
    return _run_ac_pf_without_qlims_enforced(ppci, recycle, makeYbus, ppopt)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\pf\runpf_pypower.py", line 135, in _run_ac_pf_without_qlims_enforced
    ppci, Ybus, Yf, Yt = _get_Y_bus(ppci, recycle, makeYbus, baseMVA, bus, branch)

  File "C:\Users\zhech\Anaconda3\lib\site-packages\pandapower\pf\runpf_pypower.py", line 122, in _get_Y_bus
    if recycle is not None and not recycle["trafo"] and ppci["internal"]["Ybus"].size:

TypeError: 'bool' object is not subscriptable

rbolgaryn, 2020-12-06:
This is connected to the "recycle" issue - #1011 

It is a bug that sometimes occurs, and we need to fix it.
Try to set the parameter "recycle" to False explicitly when you create controllers. If it doesn't help, try setting it to None.

rbolgaryn, 2020-12-06:
please provide a small example so that we can use it to fix the bug, because it is not very clear yet how it happens

rbolgaryn, 2020-12-06:
#1040

rbolgaryn, 2020-12-07:
Hi @hitchenzhe ,

can you pull pandapower and try again?

closed 2020-12-10 by rbolgaryn