e2nIEE/pandapower #1041: 3 phase load flow does not converge for grid with wye-connected unbalanced loads

ishoukry, 2020-12-07:
Hi there

I have created a grid to test the three phase power flow algorithm of pandapower with and have run into an issue.

I created a grid with 4 buses, 1 Transformer and 2 lines. When I add a **wye connected** load on one of the phases, the algorithm **does not converge**. When I add the same load but **delta connected** the algorithm **converges** with realistic looking results.

Any idea what might be wrong here?

Here is the code for reproduction:
`import pandapower as pp

# empty grid
grid = pp.create_empty_network() 

# define unsymmetric line as standard type (component library)
unsymmetric_line = {"r_ohm_per_km": 0.1941, "x_ohm_per_km": 0.07476991, "c_nf_per_km": 1160., "max_i_ka": 0.421,"endtemp_degree": 70.0, "r0_ohm_per_km": 0.7766,"x0_ohm_per_km": 0.2990796,"c0_nf_per_km":  500.}
pp.create_std_type(grid, unsymmetric_line, name="unsymmetric_line",element = "line")

# define unsymmetric trafo as standard type (component library)
unsymmetric_trafo = {"sn_mva": 1.6, "vn_hv_kv": 20,"vn_lv_kv": 0.4, "vk_percent": 6, "vkr_percent": 0.78125, "pfe_kw": 0.0, "i0_percent": 0.0, "shift_degree": 0, "vector_group": "Dy", "tap_side": "lv", "tap_neutral": 0, "tap_min": -2, "tap_max": 2, "tap_step_degree": 0, "tap_step_percent": 2.5, "tap_phase_shifter": False, "vk0_percent": 6, "vkr0_percent": 0.78125, "mag0_percent": 100, "mag0_rx": 0., "si0_hv_partial": 0.9}
pp.create_std_type(grid,unsymmetric_trafo, name='unsymmetric_trafo', element="trafo")

# create buses
b1 = pp.create_bus(grid, vn_kv=20.0, name="Bus 1")
b2 = pp.create_bus(grid, vn_kv=0.4, name="Bus 2")
b31 = pp.create_bus(grid, vn_kv=0.4, name="Bus 3.1")
b32 = pp.create_bus(grid, vn_kv=0.4, name="Bus 3.2")

# create network feeder
pp.create_ext_grid(grid, bus=b1, vm_pu=1.02, name="Grid Connection", s_sc_max_mva=1000, rx_max=1, x0x_max=1, r0x0_max=1)

# create loads
# pp.create_load(grid, bus=b31, p_mw=0.01, q_mvar=0, const_z_percent=0, const_i_percent=0, sn_mva=0.01, name=None, scaling=1.0, index=None, in_service=True, type='wye', max_p_mw=0.0, min_p_mw=0.0, max_q_mvar=0.0, min_q_mvar=0.0, controllable=False);
pp.create_asymmetric_load(grid, bus=b31, p_a_mw=0.0, p_b_mw=0.0, p_c_mw=0.1, q_a_mvar=0, q_b_mvar=0, q_c_mvar=0, sn_mva=0.6, name=None, scaling=1.0, index=None, in_service=True, type='wye')
pp.create_asymmetric_load(grid, bus=b31, p_a_mw=0.0, p_b_mw=0.0, p_c_mw=0.8, q_a_mvar=0, q_b_mvar=0, q_c_mvar=0, sn_mva=0.06, name=None, scaling=1.0, index=None, in_service=True, type='delta')

# create trafos
t1 = pp.create_transformer(grid, hv_bus=b1, lv_bus=b2, std_type="unsymmetric_trafo", name="Trafo")

# create lines
l1 = pp.create_line(grid, from_bus=b2, to_bus=b31, length_km=0.1, name="Line",std_type="unsymmetric_line") 
l2 = pp.create_line(grid, from_bus=b2, to_bus=b32, length_km=0.1, name="Line",std_type="unsymmetric_line") 
# l3 = pp.create_line(grid, from_bus=b31, to_bus=b32, length_km=0.1, name="Line",std_type="unsymmetric_line") 

# adds the zero sequence parameters of the standard types to the grid model
pp.add_zero_impedance_parameters(grid)

# run three-phase power flow
pp.runpp_3ph(grid, calculate_voltage_angles=True, init='auto', max_iteration='auto', tolerance_mva=1e-08, trafo_model='t', trafo_loading='current', enforce_q_lims=False, numba=True, recycle=None, check_connectivity=True, switch_rx_ratio=2.0, delta_q=0, v_debug=False)
`

lthurner, 2020-12-15:
@gourab2009 can you look into this?

gourab2009, 2020-12-15:
Hi Leon,

I'll go back to work tomorrow, and have a look on the issue.

Regards,
Gourab

gourab2009, 2020-12-16:
Hi @ishoukry 
Let me answer to the problem you are facing:

-There is a little correction required to the creation of "unsymmetric_trafo".
- Pandapower unbalanced power flow supports " ‘Dyn’,’Yyn’,’Yzn’,’YNyn’ " vector group. 
- Change from ""vector_group": "Dy" to "vector_group": "Dyn", and then it will work and converges perfectly.

Just to explain to you the reason why it still worked for "delta" load is that the delta connection blocks zero-sequence current flow, and didn't impact. Pandapower unbalanced power flow doesn't support floating neutral point, and calculation related to that as it's also not a relevant used case for LV grid. Neutral is always ground connected in LV grids.

One more suggestion: Please check p_mw and sn_mva values as well while you are creating asym._loads.

Regards,
Gourab

closed 2020-12-16 by rbolgaryn

ishoukry, 2020-12-17:
Thanks @gourab2009 for the quick clarification. I was hoping it would be a configuration issue of the grid model. I tried it out with the changed vector group and it works as expected.

Regarding your comment about p and s_n. Is it expected that s_n is the sum of the three values entered for p_a, p_b and p_c?