e2nIEE/pandapower #1046: preserve dtypes when adding JSONSerializables to net

dlohmeier, 2020-12-08:
In the basic controller, dtypes are preserved upon creation. I am wondering if this is something that should be shifted to the JSONSerializableClass when an object is added to a net, because this resembles a create-function. For other tables with that kind of behavior, this would have to be implemented as well.

SteffenMeinecke, 2021-11-11:
With https://github.com/SteffenMeinecke/pandapower/commit/6e7f9262fbbf5eee6ce218a7bd84601d5fc4e38d this should be fixed at least for the loading of files (FileIO). For creating entries to such object tables, this issue still exists.

dlohmeier, 2021-11-15:
Sorry, I forgot to add a code snippet. It would be added to the JSONSerializableClass method `add_to_net`:

Before:

    def add_to_net(self, net, element, index, column="object", overwrite=False):
        if element not in net:
            net[element] = pd.DataFrame(columns=[column])
        if index in net[element].index.values:
            obj = net[element].object.at[index]
            if overwrite or not isinstance(obj, JSONSerializableClass):
                logger.info("Updating %s with index %s" % (element, index))
            else:
                raise UserWarning("%s with index %s already exists" % (element, index))

Afterwards:

    def add_to_net(self, net, element, index, column="object", overwrite=False, preserve_dtypes=False):
        if element not in net:
            net[element] = pd.DataFrame(columns=[column])
        if index in net[element].index.values:
            obj = net[element].object.at[index]
            if overwrite or not isinstance(obj, JSONSerializableClass):
                logger.info("Updating %s with index %s" % (element, index))
            else:
                raise UserWarning("%s with index %s already exists" % (element, index))
        if preserve_dtypes:
            dtypes = net[element].dtypes
            net[element].at[index, column] = self
            _preserve_dtypes(net[element], dtypes)
        else:
            net[element].at[index, column] = self

dlohmeier, 2021-11-15:
This way, inheriting classes could make use of this functionality without having to take care of this for each other class, just by setting a flag.

PR #1412 (2021-12-02): preserve dtypes possible when adding json serializable to net
closes #1046
commit 411d7a00 (2021-12-02): preserve dtypes possible when adding json serializable to net (closes #1046)
closed 2021-12-02 by dlohmeier