e2nIEE/pandapower #1050: runpp with numba=False still uses numba

rbolgaryn, 2020-12-08:


rbolgaryn, 2020-12-08:
```python
try:
    get_values = jit(nopython=True, cache=True)(_get_values)
    set_elements_oos = jit(nopython=True, cache=True)(_python_set_elements_oos)
    set_isolated_buses_oos = jit(nopython=True, cache=True)(_python_set_isolated_buses_oos)
except RuntimeError:
    get_values = jit(nopython=True, cache=False)(_get_values)
    set_elements_oos = jit(nopython=True, cache=False)(_python_set_elements_oos)
    set_isolated_buses_oos = jit(nopython=True, cache=False)(_python_set_isolated_buses_oos)
```

wangzhenassd, 2021-01-19:
@vogt31337 Is this the problem you encountered?

vogt31337, 2021-02-07:
```
Traceback (most recent call last):
  File "start_opf_mike.py", line 1010, in calculate_timestep
    pp.runpp(net, run_control=True, trafo_taps=True, numba=False, max_iter=100, max_iteration=100, tolerance_mva=1e-6)
  File "/home/vogt/pandapower/pandapower/run.py", line 213, in runpp
    run_control(**parameters)
  File "/home/vogt/pandapower/pandapower/control/run_control.py", line 290, in run_control
    ctrl_variables = net_initialization(net, ctrl_variables, **kwargs)
  File "/home/vogt/pandapower/pandapower/control/run_control.py", line 162, in net_initialization
    run_funct(net, **kwargs)  # run can be runpp, runopf or whatever
  File "/home/vogt/pandapower/pandapower/run.py", line 227, in runpp
    _powerflow(net, **kwargs)
  File "/home/vogt/pandapower/pandapower/powerflow.py", line 66, in _powerflow
    ppc, ppci = _pd2ppc(net)
  File "/home/vogt/pandapower/pandapower/pd2ppc.py", line 62, in _pd2ppc
    net["_is_elements"] = aux._select_is_elements_numba(net, sequence=sequence)
  File "/home/vogt/pandapower/pandapower/auxiliary.py", line 476, in _select_is_elements_numba
    set_elements_oos(element_df["bus"].values, element_df["in_service"].values,
TypeError: can't unbox array from PyObject into native value.  The object maybe of a different type
```

Still persist @wangzhenassd and is the main cause of fails for 2 of 3 runs. So please fix it, because it is really annoying.

wangzhenassd, 2021-02-07:
@vogt31337 can you provida a piece of code to reproduce this?

wangzhenassd, 2021-02-07:
@vogt31337 just send me the network, or I cannot start

vogt31337, 2021-02-07:
Unfortnately I cannot. Firstly the model is only for internal use, secondly this problem occurs on parallelized calculations on a cluster. So I think the best way would be to check if numba is really disabled when "numba = False" is selected. Since I think it isn't correctly implemented on some lower levels...

wangzhenassd, 2021-11-11:
I am not sure, if I can find time on this after 03.2022.

vogt31337, 2025-03-31:
closed due to inactivity.

closed 2025-03-31 by vogt31337