e2nIEE/pandapower #1064: Question on shortcircuit test

wangzhenassd, 2020-12-15:
```
    net = pp.create_empty_network(sn_mva=2.)
    b0 = pp.create_bus(net, 220)
    b1 = pp.create_bus(net, 110)
    b2 = pp.create_bus(net, 110)
    b3 = pp.create_bus(net, 110)
    pp.create_ext_grid(net, b0, s_sc_max_mva=100., s_sc_min_mva=80., rx_min=0.4, rx_max=0.4)
    pp.create_transformer(net, b0, b1, "100 MVA 220/110 kV")
    pp.create_line(net, b1, b2, std_type="305-AL1/39-ST1A 110.0" , length_km=20.)
    l2 = pp.create_line(net, b2, b3, std_type="N2XS(FL)2Y 1x185 RM/35 64/110 kV" , length_km=15.)
    pp.create_line(net, b3, b1, std_type="N2XS(FL)2Y 1x185 RM/35 64/110 kV" , length_km=10.)
    pp.create_switch(net, b3, l2, closed=False, et="l")
    sc.calc_sc(net, branch_results=True, inverse_y=False)
    assert np.allclose(net.res_trafo_sc.ikss_lv_ka.values, [0.47705988])
    assert np.allclose(net.res_line_sc.ikss_ka.values, [0.45294928, 0.44514686, 0.47125418])
```
This is the first test in test_ring.py. On line 2 the switch is open, I think the sc current on a line with opened switch should be zero. Why is it here 0.445? Am I missing something here?

wangzhenassd, 2020-12-15:
@lthurner @rbolgaryn

rbolgaryn, 2020-12-15:
Seems logical that line 2 should have 0

rbolgaryn, 2020-12-15:
I tried a similar configuration in PowerFactory, and the short-circuit current of a line with an open switch is 0

wangzhenassd, 2020-12-15:
@rbolgaryn that's interested to know. I think pandapower can follow this convention maybe set result of out-of-service element to 0 or do you think it makes more sense to set as NaN?

rbolgaryn, 2020-12-15:
I think it is not out of service, so then it should be 0

lthurner, 2020-12-16:
Without having checked, my guess is that the reason for this behaviour is that a short-circuit will be calculated for the auxiliary bus of the open switch as well. I agree the behaviour is not intuitive, would be better to set it to zero.

I agree it should be 0 instead of NaN.

PR #1082 (2021-01-28): Update short circuit calculation 
According to #1037, the performance of short circuit for large networks are optimized:
1. added LU factorization mode instead of complete inversion of Ybus 
2. added bus selection and updated the result checkout
3. updated some tests
4. fixed #1004 
5. fixed #1064
6. Accroding to #78, a part of the example in iec60909 is now added
closed 2021-01-28 by lthurner