e2nIEE/pandapower #1065: Bug in OPF init

thomaswolgast, 2020-12-16:
I think there is quite an obvious bug in the OPF calculation:

The `_add_ppc_options `method can only get `init='pf'` or `init='flat'` (see [Dokumentation](https://pandapower.readthedocs.io/en/v2.4.0/opf/pypower_run.html#ac-opf)). However, the method expects  [`init_vm_pu=="results"`](https://github.com/e2nIEE/pandapower/blob/156472ff28dd7d8d1c742da4c5e96f5b7199e3ec/pandapower/auxiliary.py#L503) (whereas `init_vm_pu `was set to `init` [before](https://github.com/e2nIEE/pandapower/blob/156472ff28dd7d8d1c742da4c5e96f5b7199e3ec/pandapower/auxiliary.py#L1044)). 

I assume the ´pf´ option was called ´results´ in the past and that line was never changed? Because this way, the condition will always be wrong, since there is no option to set ´init='results'´. If that is correct, the conditions in [this line](https://github.com/e2nIEE/pandapower/blob/156472ff28dd7d8d1c742da4c5e96f5b7199e3ec/pandapower/auxiliary.py#L503) should be changed to `init_vm_pu=="pf"`.

That would be easy to change, but I don't know if there are consequences for other parts of the code.

friederikemeier, 2020-12-16:
@thomaswolgast You're right. I think this is an artifact. We would highly appreciate a pull request on this 😄

rbolgaryn, 2020-12-16:
I can see that the handlöing of "init" has many if-clauses in the function that sets options for pp.runpp, but not for opf.

thomaswolgast, 2020-12-16:
The problem I see here is that ´_add_ppc_options()´ is not only used for the OPF, but also for the normal powerflow calculation. And there `init="results"` is perfectly valid. Therefore, we cannot simply change the if-condition to `init="pf"`. Is there an actual difference between `init="results"` of the powerflow calc and `init="pf"` of the OPF calc? In both cases, the last powerflow calculation is used to set the start values of the calculation, right? So why differentiate between both cases?

friederikemeier, 2020-12-17:
Actually I remember now, that there is/was a difference.. "pf" should trigger an initial pp.runpp before starting the opf itself. "results" should take the results stored in the pandapower network (res_bus.vm_pu etc.) as an initialization. If we use "results", those results used for initialization do not necessarily originate from a runpp, they could also be results of a previous runopp... I remember that it was necessary to distinguish between those cases.

closed 2021-07-21 by friederikemeier