e2nIEE/pandapower #1067: KeyError: 'continue_on_divergence' using run_control

SteffenMeinecke, 2020-12-16:
Using the following code, the attached error occur. It seems that with some changes in the code, the key `continue_on_divergence` got lost.

```
import pandapower as pp
from pandapower.control import run_control

net = pp.from_json("testnet.json")
run_control(net, calculate_voltage_anlges=True, max_iter=100)
```
[image]
[testnet.zip](https://github.com/e2nIEE/pandapower/files/5702349/testnet.zip)

SimonRubenDrauz, 2021-01-11:
should be solved. Could you check it!

rbolgaryn, 2021-02-23:
Hi @SteffenMeinecke , can the Issue be closed?

SteffenMeinecke, 2021-03-01:
No, nothing has been done yet :(

closed 2021-03-01 by SteffenMeinecke