e2nIEE/pandapower #1072: false power flow results for simbench and MV Oberrhein - 3ph power flow

eva-bct, 2020-12-18:
Problem: 
because type of "load" and "sgen" in simbench networks and MV Oberrhein network is not "wye" or "delta", the algorithm runpp_3ph set these elements internaly to non active
--> wrong power flow results

in runpp_3p 
line 466 --> _load_mapping 
line 139 --> _get_elements
line 80:
```active = (net["_is_elements"][element]) & (elm[:,typ_col] == typ)```

for internal calcualtion an element is „active“, if:
1) "_is_elements“==True (element in operation, information from ppci structure)
and
2) „type“ == „wye“ or „delta“ 

otherwise the elements were set to "non active"

For simbench net the "type" of "load" and "sgen" is "Nan" --> so all loads and sgen are set to non active
For MV Oberrhein the "type" of "load" is "MV/LV Station" and the "type" of "sgen" is "PV" --> all loads and sgen are set to non active

this can not be seen in ppci (here they are in operation "_is_elements" ==True) and in result_tables it can not be seen either, because they were overwritten by given load/feed data at "PQ" points

Solution:
* [ ] check before, if "type" is set to "wye" or "delta"
* [ ] if not: give a warning or set it by default to "wye"

For example:
```
load_elements = ['load', 'asymmetric_load', 'sgen', 'asymmetric_sgen']
load_types = ['wye', 'delta']

for elm in load_elements:
    for idx in NET[elm].index:
        if NET[elm].type.loc[idx] in load_types:
            continue
        else:
            NET[elm].type.loc[idx] = 'wye'
```

rbolgaryn, 2021-03-04:
Hi @eva-bct ,

symmetric loads are ignored in the unbalanced load flow at the moment, it is "work in progress" to represent them internally as 3 1ph loads

eva-bct, 2021-03-04:
Hi,

this is not the problem. 
The problem accures due to an unset load type or false set load type. 
To suceed 3 ph load flow with networks, which are initially have no correct load type (wye or delta) there has to be an query. 
When the loads are set manually to any of the both types the 3 phase load works properly.

rahatN-tech, 2023-06-09:
i have been trying to run pf with backward forward sweep on simbench 1-LV-rural1--0-sw network
i have also corrected the slack bus index from 42 to 14
but i am getting the following error

File "d:/python rahat/osman boyaci/pp_try_four.py", line 37, in <module>
    pp.runpp(net, algorithm='bfsw')
  File "C:\Users\Mairah\AppData\Local\Programs\Python\Python37\lib\site-packages\pandapower\run.py", line 232, in runpp
    _powerflow(net, **kwargs)
  File "C:\Users\Mairah\AppData\Local\Programs\Python\Python37\lib\site-packages\pandapower\powerflow.py", line 84, in _powerflow
    result = _run_pf_algorithm(ppci, net["_options"], **kwargs)
  File "C:\Users\Mairah\AppData\Local\Programs\Python\Python37\lib\site-packages\pandapower\powerflow.py", line 162, in _run_pf_algorithm
    result = _run_bfswpf(ppci, options, **kwargs)[0]
  File "C:\Users\Mairah\AppData\Local\Programs\Python\Python37\lib\site-packages\pandapower\pf\run_bfswpf.py", line 408, in _run_bfswpf
    ppci, DLF, buses_ordered_bfs_nets = _get_bibc_bcbv(ppci, options, bus, branch, G)
  File "C:\Users\Mairah\AppData\Local\Programs\Python\Python37\lib\site-packages\pandapower\pf\run_bfswpf.py", line 178, in _get_bibc_bcbv
    DLF, buses_ordered_bfs_nets = _make_bibc_bcbv(bus, branch, graph)
  File "C:\Users\Mairah\AppData\Local\Programs\Python\Python37\lib\site-packages\pandapower\pf\run_bfswpf.py", line 149, in _make_bibc_bcbv
    shape=(nobranch, nobranch))
  File "C:\Users\Mairah\AppData\Local\Programs\Python\Python37\lib\site-packages\scipy\sparse\compressed.py", line 54, in __init__
    other = self.__class__(coo_matrix(arg1, shape=shape))
  File "C:\Users\Mairah\AppData\Local\Programs\Python\Python37\lib\site-packages\scipy\sparse\coo.py", line 196, in __init__
    self._check()
  File "C:\Users\Mairah\AppData\Local\Programs\Python\Python37\lib\site-packages\scipy\sparse\coo.py", line 289, in _check
    raise ValueError('negative column index found')
ValueError: negative column index found

plz help
badly stuck

SteffenMeinecke, 2023-06-12:
Dear @rahatN-tech,
your issue does not fit to the topic of this issue. Please open a new issue and link to this issue so one can find my first answer:
I can reproduce your issue with 
```
import pandapower as pp
import simbench as sb
net = sb.get_simbench_net("1-LV-rural1--0-sw")
pp.runpp(net, algorithm="bfsw")
```
Is it mandatory for you to use the backward-forward sweep algorithm?
If not, you can just use the default algorithm which works fine.
To find out the reason for the bfsw crash someone is needs to take a look into the code.

rahatN-tech, 2023-06-12:
This is a low level radial distribution system.. So backward forward sweep
algorithm is better suited for it
This algorithm is nt working on any simbench network. ۔though it is working
fine on pandapower sample networks

Kindly help

؟