e2nIEE/pandapower #1073: transformer losses in 3ph power flow three times higher than in 1ph power flow

eva-bct, 2020-12-18:
comparing the results from 3phase power flow and single phase power flow little differences (0.6-1.36%) occure between the results 
(compare p_mw from net.res_bus with sum of p_a_mw, p_b_mw and p_c_mw from net.res_bus_3ph)
they are probably caused by too high transformer loss settings in 3 ph power flow

when 
1) first running single phase power flow
2) than changing transformer settings:
```
NET.trafo.pfe_kw = NET.trafo.pfe_kw/3
NET.trafo.i0_percent = NET.trafo.i0_percent/3
```
3) afterwards running three phase power flow

the differences between the results are now less than 2 e-11 %

rbolgaryn, 2020-12-18:
Hi @gourab2009 , can you please take a look?

gourab2009, 2021-02-15:
Hello,
Thanks for your observation @eva-bct.
I have checked the issue and your suggestion. It seems to be giving a better result when the iron loss is divided by 3.
We will look into it further with a complex network and modify the calculation accordingly.

lthurner, 2021-04-12:
Thanks @eva-bct for reporting and @elodin for fixing. Closed by #1201

closed 2021-04-12 by lthurner