e2nIEE/pandapower #1083: Topology flag in short circuit

SimonRubenDrauz, 2021-01-05:
Few days ago I was checking out the pandapower tutorials and I think I found a bug while doing them. 

In the shortcircut tutorial I found out that when conducting a short circuit calculation the topology flag doesn't seem to work properly, Have a look at following screensshot:
[image]
No matter if I type topology='auto' or topology='radial' the results don't change but stay the same even though that shouldn't be the case. 

Could you have a look, @lthurner?

rbolgaryn, 2021-02-23:
The switch gets closed properly, and yet the results are the same

SimonRubenDrauz, 2021-02-23:
Is this a question or do you agree with me?

rbolgaryn, 2021-02-23:
I agree with you

closed 2025-11-26 by heckstrahler