e2nIEE/pandapower #1085: Strange behavior in some cases of plotting branch elements

dlohmeier, 2021-01-07:
I just found that there is a line of code that might really lead to wrong plotting output in some cases. In the function `coords_from_node_geodata` used for all branch elements without their own geodata, the coordinates are only created for those elements where from_node and to_node don't have the same geodata. Especially in the case of switches, this might cause problems for some network models, as for some switching stations, all buses have the same geodata. I know that I copied these lines of code from some other function some time ago, but I think that this is undesired behavior, or is there a problem in creating line collections containing lines with the same start and end point?

SteffenMeinecke, 2021-11-11:
Obviously, no one has described similar problems.

> might really lead to wrong plotting output in some cases

@dlohmeier Does it definitely lead to problems? If yes, could you provide an example code? If not, I would close this issue and reopen it if some finds similar problems in future.

dlohmeier, 2021-11-15:
Well, I had an issue with some data which I can't recall, but I was able to handle it by adding a small flag to that function. I think, it's all right to reopen this issue, if someone experiences problems again.

closed 2021-11-15 by dlohmeier