e2nIEE/pandapower #1094: run time series not possible with multiple loads at same bus

eva-bct, 2021-01-08:
take a network, which has multiple loads at one bus, the output writer in ```run_time_series``` fails 

the problem occurs with json file export: \
the column names of the dataframe ```net.res_bus``` are the bus index to which each load is connected \
so if more loads are connected to the same bus, there are columns with the same name 

This does not work, because the columns have to be unique for panda json file export:

```ValueError: DataFrame columns must be unique for orient='columns'.```

eva-bct, 2021-01-08:
sometimes it is necessary to have two or more loads connected to one bus as well as different infeed sources (sgen)

SimonRubenDrauz, 2021-01-11:
Hi Eva,

would it be possible to share your network with us in order to understand what might be wrong. I tried to reproduce your problem but I was not able to do so. What I did:
- At first I loaded a network from pandapower. I loaded the case4gs.json (net = pp.from_json(filename)).
- Afterwards I created a new load with following comment: pp.create_load(net, 1, 300). At bus 1 another load was already installed. The net looks as follows:
[image]
- Then I imported run_timeseries by 'from pandapower.timeseries import run_timeseries'.
- Finally I conducted a timeseries simulation --> run_timeseries(net, range(20)). Everything seems fine.
- In order to find out if it might be a saving-problem, I also did pp.to_json(net, 'test.json) and net2 = pp.from_json('test.json').
- Afterwards I did run_timeseries(net2, range(20)) again.
So far I couldn't find any problems doing it so.  

If I misunderstood your problem, looking forward for your clarification.

Beste regards,

Simon

eva-bct, 2021-01-11:
Hi Simon,

Yes, of course, \
I did nearly the same procedure as you did:

-  I loaded a network in pandapower (simbench network ```1-LV-rural1--0-sw```) and added an additional load to bus 5
- the network looks like: 
[image]

-  Afterwards initialized ConstControl and OutputWriter: ```ow = OutputWriter(NET, time_steps, output_path=output_dir, output_file_type=".json") ``` \
with log variables, e.q.: ```ow.log_variable('res_bus_3ph', 'vm_a_pu', index=net.load.index)```

- then run time series with three phase power flow
```
run_timeseries(
    net,
    time_steps=time_steps,
    run=pp.runpp_3ph,
    continue_on_divergence=True,
    verbose=True,
)
```

- following message appears:
[image]

- for no multiple loads at one bus, the code works properly 

Thanks a lot for helping!

Kind regards,
Eva

eva-bct, 2021-01-12:
I think I found my mistake:
It is not possible to give the log variable an index list with multiple loads.

I am sorry for bothering you with this!

closed 2021-01-12 by eva-bct

SimonRubenDrauz, 2021-01-12:
cool, happy, that it is solved. No problem. 

Another thought: What you could also try: Actually you shouldn't pass the load_index, but the bus_index, otherwise it is using the load indices in the res_bus DataFrame. You could try:
ow.log_variable('res_bus_3ph', 'vm_a_pu', index=net.bus.index) or ow.log_variable('res_bus_3ph', 'vm_a_pu', index=set(net.load.bus.values). This should hopefully also solve your problem with multiple loads.

eva-bct, 2021-01-12:
Exactly, I used "set", and it works perfectly now