e2nIEE/pandapower #1098: Unbalanced power flow - impedance

AnBlav, 2021-01-12:
Hello,

I would like to simulate an unbalanced power flow with a realistic voltage source (i.e. having a non-zero series impedance in order to model the strength of the grid). In balanced power flow calculations, I use the external grid in series with the impedance element to model the grid strength. However, as the impedance element is not implemented yet for unbalanced power flows, I have been recommended to use the external grid for this and more specifically its parameters for short-circuit calculations: s_sc_max_mva, rx_max, r0x0_max, x0x_max.

The impedance included in the external grid is supposed to be balanced. I assume that the short-circuit level is to be entered for parameter s_sc_max_mva,  and parameter rx_max helps defining the impedance angle. Parameter r0x0_max is also self-explanatory. Here are my questions:

- I suppose that I should leave s_sc_min_mva and rx_min empty?

- Why is parameter x0x limited to 1? Alternatively, could you describe the meaning of "x" (positive/negative sequence component? etc.).

Thanks in advance for your answers.

AnBlav, 2021-01-29:
Hi,

I haven't received any answer to my question. Could you please help me?

Best regards,

Anne

wangzhenassd, 2021-01-29:
HI,
I think s_sc_max_mva, rx_max, r0x0_max, x0x_max is only related for short circuit calculation as defined in IEC 60909. What you can possibly do, is to use a small piece of line with the expected unbalanced impedance value.

SteffenMeinecke, 2021-11-11:
@AnBlav Hopefully the comment helped you. As no further questions were raised, I close this issue.

closed 2021-11-11 by SteffenMeinecke