e2nIEE/pandapower #1100: Issue Tutorial opf_dcline.ipynb. pp.runopp: Optimal Power Flow did not converge!

ballesm, 2021-01-13:
Greetings, 
I downloaded pandapower 2.5.0 with Python 3.8.6.
I am running tutorial opf_dcline.ipynb.
When I run lines

import pandapower as pp
from numpy import array
net = pp.create_empty_network()

b1 = pp.create_bus(net, 380)
b2 = pp.create_bus(net, 380)
b3 = pp.create_bus(net, 380)
b4 = pp.create_bus(net, 380)
b5 = pp.create_bus(net, 380)

l1 = pp.create_line(net, b1, b2, 30, "490-AL1/64-ST1A 380.0")
l2 = pp.create_line(net, b3, b4, 20, "490-AL1/64-ST1A 380.0")
l3 = pp.create_line(net, b4, b5, 20, "490-AL1/64-ST1A 380.0")

dcl1 = pp.create_dcline(net, name="dc line", from_bus=b2, to_bus=b3, p_mw=200, loss_percent=1.0, 
                  loss_mw=0.5, vm_from_pu=1.01, vm_to_pu=1.012, max_p_mw=1000,
                  in_service=True)

eg1 = pp.create_ext_grid(net, b1, 1.02, min_p_mw=0.)
eg2 = pp.create_ext_grid(net, b5, 1.02, min_p_mw=0.)

l1 = pp.create_load(net, bus=b4, p_mw=800, controllable = False)

costeg0 = pp.create_poly_cost(net, 0, 'ext_grid', cp1_eur_per_mw=10)
costeg1 = pp.create_poly_cost(net, 1, 'ext_grid', cp1_eur_per_mw=8)
net.bus['max_vm_pu'] = 1.5
net.line['max_loading_percent'] = 1000

pp.runopp(net)

the following error arises.  

Traceback (most recent call last):

  File "C:\Ancillary_Analysis\PYTHON\Pandapower\Code\opf_dcline.py", line 29, in <module>
    pp.runopp(net)

  File "C:\WPy64-3860\python-3.8.6.amd64\lib\site-packages\pandapower\run.py", line 357, in runopp
    _optimal_powerflow(net, verbose, suppress_warnings, **kwargs)

  File "C:\WPy64-3860\python-3.8.6.amd64\lib\site-packages\pandapower\optimal_powerflow.py", line 70, in _optimal_powerflow
    raise OPFNotConverged("Optimal Power Flow did not converge!")

OPFNotConverged: Optimal Power Flow did not converge!

I do not change any lines of code from the published tutorial. All other tutorials work, just dc lines do not converge. Do you have any idea of the reason? I did not find any solutions and pp.diagnostic(net) find no issue
Thank you

renato-monaro, 2021-02-17:
I am facing the same problem. I tried to run opf for a custom system and got a converge error. Then I tried the example opf_dcline to check if the error was on my side, but I got converge error.

friederikemeier, 2021-07-21:
This is really strange, as it was running before. I don't have the time to check this myself. But I suggest to run the code with an older version of pandapower, where it was still working. And then extract the net._ppc and compare it to the net._ppc you are getting with the actual version. What are the differences there? 
I guess we introduced a change in pd2ppc, that we didn't test with dc_line and OPF...

heckstrahler, 2025-11-24:
When using the code from the tutorial, runopp runs successfully, delta is needed.
pp.runopp(net, delta=1e-16)

closed 2025-11-24 by heckstrahler