e2nIEE/pandapower #1101: Pandapower OPF does not converge with pp version > 2.2.0

joshuajakob, 2021-01-14:
Hello,
I have some problems with pandapower after updating to a version higher 2.2.0.
Same code, works fine on pandapower 2.2.0 but on version 2.3.0 or even 2.5.0 the "Optimal Power Flow did not converge!"

################################## example code #################################################
net = sb.get_simbench_net("1-MV-semiurb--0-sw")
pp.create_sgen(net, bus=25,p_mw = 0,name = 'KWK', controllable = True,min_p_mw=0, max_p_mw=100, max_q_mvar = 0, min_q_mvar = 0, index = 1000)
pp.create_poly_cost(net, 1000, 'gen', cp1_eur_per_mw=-1)

pp.runopp(net)
##############################################################################################

It would be great if you could help me.
Thanks in advance!

rbolgaryn, 2021-02-24:
Hi @friederikemeier , can you please take a look?

ricrei, 2021-07-13:
Hello,
I run into the same problem. The above-mentioned code works with pp 2.2.0. But I had to change the 'gen' to 'sgen' in the cost function. With pp >2.2.0 it does not converge like other simbench grids or the provided kerber grids. That is strange because the OPF tutorials work fine even with pp >2.2.0.
It would be great if you could take a look.
Best!

ivanightingale, 2021-08-04:
runopp does not converge for case5

friederikemeier, 2021-08-04:
As far as I remember, case5 was never converging. This must be a different issue.
In Matpower/Pypower, case5 is converging, as it is already initiliazed in the optimal state.

salvapineda, 2021-11-16:
I am also trying to solve the acopf for case5() and it is not converging either. Do you have any idea about why this is happening? 
Thanks a lot!

thomaswolgast, 2023-07-31:
Hello, the problem with the non-converging OPF in pandapower>=2.3.0 seems to be that the default for `calculate_voltage_angles` was changed from False to True (see https://pandapower.readthedocs.io/en/v2.3.0/about/changelog.html). 
When you set `calculate_voltage_angles=False`, the OPF should work again. Why the OPF is failing when you want to compute voltage angles, I don't understand either. However, this is a quick solution to have a working OPF again.