e2nIEE/pandapower #1108: GitHub Actions: linting seems to run tests

dlohmeier, 2021-01-19:
I just created a new PR and had a look at the checks. The linting step seems to take very long, and it appeared to me that it performed all tests in Python 3.8, although this should be included in the build step. Is it true that this action performs all the tests, or does it only have to wait for the build to finish? I think it wouldn't make sense to run all tests again for the coverage report, otherwise it should be included in the build action.

jkisse, 2021-02-01:
I think this is not related to the linting step (< 30 s), but to the coverage report that had been added to the same job (> 8 min).
Is a coverage report even possible without running all tests? Isn't that basically what the report is about? It would make sense though to combine the coverage report with the regular Python 3.8 test.

jkisse, 2021-02-01:
In addition, we have the Codecov Integration that checks the coverage, so we could remove the coverage report from GitHub Actions.

PR #1128 (2021-02-03): Run tests for 3.8 with codecov
Remove codecov from flint, run 3.8 with codecov, closes #1108
commit c14bd954 (2021-02-03): Run tests for 3.8 with codecov

Remove codecov from flint, run 3.8 with codecov, closes #1108
closed 2021-02-03 by rbolgaryn