e2nIEE/pandapower #1115: Successfully build environments for Pyjulia. May contribute to Powermodels OPF tutorials

lvzhibai, 2021-01-21:
I have previously encountered PyJulia problems when using anaconda Python 3.8.5. The PyJulia never works with any solutions online.

Here is my solutions. Solved in MacOS 11.0, Python 3.9.1

1. Follow [Powermodels tutorial](https://pandapower.readthedocs.io/en/v2.5.0/opf/powermodels.html) step 1 to set julia. Also install PyCall using `]add PyCall`
2. Download original python program (different version is better) from python.org and install.
3. In terminal, run specific command for this python (in my case `python3.9`) and check if it doesn't show anaconda.
4. Use command `python3.9 -m pip install foo bar ...` to install pandapower, julia and required packages.
5. Inside this python, run
```
>>> import julia
>>> julia.install() 
```
This will set PyJulia and PyCall correctly.
6. Follow step 3,4,6,7 of [Powermodels tutorial](https://pandapower.readthedocs.io/en/v2.5.0/opf/powermodels.html). The test will pass.

rbolgaryn, 2021-02-23:
Hi @lvzhibai ,

your contributions to tutorials are most welcome, please feel free to open a pull request.

friederikemeier, 2021-07-21:
@MMajidi137 can you check on this issue, please?

vogt31337, 2025-04-03:
closed due to inactivity.

closed 2025-04-03 by vogt31337